wpilibsuite/allwpilib #3843: C++ & Java Differ in TankDriveVolts Implementations

modelmat, 2021-12-29:
**Describe the bug**
https://github.com/wpilibsuite/allwpilib/blob/aa9dfabde2fdaf32541697de2159878faa801fa1/wpilibcExamples/src/main/cpp/examples/StateSpaceDifferentialDriveSimulation/cpp/subsystems/DriveSubsystem.cpp#L65-L69

https://github.com/wpilibsuite/allwpilib/blob/aa9dfabde2fdaf32541697de2159878faa801fa1/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/statespacedifferentialdrivesimulation/subsystems/DriveSubsystem.java#L190-L200

**To Reproduce**
Read sauce

**Expected behavior**
Not be different.

**Desktop (please complete the following information):**
 - WPILib Version: [e.g. 2021.3.1] main
 - OS: [e.g. Windows 11] n/a
 - Java version [e.g. 1.10.2] n/a
 - C++ version [e.g. 17] n/a

**Additional context**
Not sure if other examples also do this.

modelmat, 2021-12-29:
RamseteCommand seems to be the only other command that implements a function tankDriveVolts, and it also exhibits this issue.

sciencewhiz, 2021-12-29:
I don't think the additional check in Java is good as it leads to discontinuities. For example of battery is 10v and you command 11v, you get an output of 0.91. But if you command 10v you get an output of 1.0.

modelmat, 2021-12-30:
I think also the `SetVoltage()` command according to the docstring also accounts for varying battery voltages too

Starlight220, 2022-05-06:
https://github.com/wpilibsuite/allwpilib/blob/ee03a7ad3bcc156d3b9c07c590e32bca5df83537/wpilibj/src/main/java/edu/wpi/first/wpilibj/motorcontrol/MotorController.java#L29-L31

https://github.com/wpilibsuite/allwpilib/blob/ee03a7ad3bcc156d3b9c07c590e32bca5df83537/wpilibc/src/main/native/cpp/motorcontrol/MotorController.cpp#L11-L13

I think that the C++ impl is the correct one?

calcmogul, 2022-05-06:
getBatteryVoltage() and getInputVoltage() return the same thing. C++ should be using GetBatteryVoltage() so it doesn't have to cast to units::volt_t.

PR #4284 (2022-06-03): Remove unnecessary cast and normalization of voltages
Closes #3843
closed 2022-06-03 by PeterJohnson