wpilibsuite/allwpilib #3862: What happens if scheduled methods exceed allocated time using TimesliceRobot?

LudumDareDevelopment, 2022-01-04:
If I schedule a function with an allocation of 5ms using TimesliceRobot but it actually ends up taking 6ms, what will happen? Will the function be interrupted and end prematurely, or will the following functions run later due to this function exceeding its allocation?

calcmogul, 2022-01-04:
An exception is thrown, since that's a configuration error that occurs during program startup. We should add that to the doxygen/javadoc.

https://github.com/wpilibsuite/allwpilib/blob/main/wpilibj/src/main/java/edu/wpi/first/wpilibj/TimesliceRobot.java#L100-L107

calcmogul, 2022-01-04:
In that case, the next callback will run immediately after the previous one ends.

closed 2022-01-04 by LudumDareDevelopment