wpilibsuite/allwpilib #3867: Add function to reset simulation data across all devices

virtuald, 2022-01-06:
In theory one could write a function to do this now, but it would be probably simpler for a function to just exist in WPILib that does this. Would be useful for writing unit/integration tests that aren't specific to a particular robot.

Starlight220, 2022-01-06:
This does exist, I just don't remember what its name is. `HAL.initialize` might work, and between tests I think that it's more correct to shutdown and reinitialize the HAL before/after each test.

virtuald, 2022-01-06:
No, it doesn't exist.

`HAL_Shutdown` ([src](https://github.com/wpilibsuite/allwpilib/blob/4647d09b5089bf3820f571f876e54c2c37b48b99/hal/src/main/native/sim/HAL.cpp#L370)) only calls a bunch of callback functions, it doesn't actually do anything useful in a testing context.

`HAL_Initialize` ([src](https://github.com/wpilibsuite/allwpilib/blob/4647d09b5089bf3820f571f876e54c2c37b48b99/hal/src/main/native/sim/HAL.cpp#L323)) only does things the first time that it's called, so it's not useful in a testing context.

virtuald, 2022-01-07:
Something to think about here, perhaps such a cleanup function should also call hal::HandleBase::ResetGlobalHandles() as well?

agelsomini, 2022-01-18:
I did some initial investigation and I didn't see any obvious Simulation container that would make it easy to either iterate over all the simulation devices and reset them, or destroy and recreate them. 

Without an easy way to access all the devices (that I saw), my thought was that there were two options:

1. Create a new "reset all" function in every simulation device that would reset all the data for that simulation, especially in the cases where there exists more than one. This way a `HAL_ResetSimulation()` function would be just a list of ResetAll functions the same as HAL_Initialize is a set of initialization functions.  In the case of AccelerometerData there would be a new function `HALSIM_ResetAllAccelerometerData` that would call `HALSIM_ResetAccelerometerData` for every supported index.
      1. One could use the existing reset functions and instead of adding new ResetAll functions, handle iterating over the number of each device in a ResetSimulation function in HAL.ccp, but that seems like it would be prone to error and easy for new devices to be forgotten.  Though that same concern can apply to adding a ResetAll function to each simulation device.
2. Make it so that `HAL_Initialize `can be called more than once and actually do a reinit. Given that the data objects for each simulation device appear to all be declared as statics in the InitializeDeviceTypeData() functions, it doesn't make them easy to destroy/recreate. My thought was that if HAL_Initialize could be called again, we could make the InitializeData functions call reset so on every call they would get reset, instead of only on the initial call when the constructor is called for the static variables. A big downside to this option is it changes the behavior of the existing API. https://github.com/wpilibsuite/allwpilib/blob/5ba69e1af1695392ea2dbb8f5130d07bb8ea1266/hal/src/main/native/sim/mockdata/AccelerometerData.cpp#L12

I'm guessing there has to be a better way, as any of these options seem like having to change a lot of classes. But I'm new to WPILib so I'm still learning how the code is structured.

virtuald, 2022-01-19:
I don't expect this sort of thing would be able to be merged into HAL in 2022, so I'm very interested in solutions that can work without needing to modify the existing HAL library -- that way I can just copy/paste it into robotpy until it makes it into the mainline API. I need this for unit/integration testing.

From that perspective, using the existing reset functions in 1i makes a lot of sense to me.

Long term, that approach does seem error prone. A better solution would be having reset functions registered automatically when devices are created, then a reset all function would just call the registered functions.

I don't think changing HAL_Initialize's behavior is wise.

ThadHouse, 2022-01-25:
To add on, changing HAL_Initialize's behavior is not just unwise, it would be HORRIBLY breaking. That function is called all over the place to work around initialization bugs in users code. So changing that is a non starter.

virtuald, 2022-02-06:
An initial implementation of this is at https://github.com/robotpy/robotpy-hal/pull/16

virtuald, 2022-02-06:
This isn't actually enough to reset simulation completely, there are also obnoxious global singletons in other places.

virtuald, 2022-02-06:
An initial stab at some of the other global structures is at https://github.com/wpilibsuite/allwpilib/pull/4007

PR #4016 (2022-09-26): [sim] Provide function that resets all simulation data
- Fixes #3867
- Include #4015, but they can be considered separately

This works, but I don't necessarily need this merged this season if we want to defer this. Questions:

* I like `HALSIM_ResetAllSimData`, but I'm open to other names
* There were a lot of includes, so I put it into its own file. Maybe it should go into MockHooks.{cpp,h} instead?
* This is separate from resetting all handles, and I think that makes sense to stay that way
* For RobotPy, I'm going wrap this in `wpilib.simulation` along with the other global reset functions  [... cut]
commit 4ddb8aa0 (2022-09-26): [sim] Provide function that resets all simulation data (#4016)

Fixes #3867
closed 2022-09-26 by PeterJohnson