wpilibsuite/allwpilib #3874: Creating a TimedRobot instance causes stepTiming to hang

virtuald, 2022-01-07:
**Describe the bug**
Creating a TimedRobot instance causes stepTiming to hang. 

**To Reproduce**
```python
import wpilib
import wpilib.simulation

class MyRobot(wpilib.TimedRobot):
    pass

robot = MyRobot()

wpilib.simulation.pauseTiming()
wpilib.simulation.stepTiming(1)

```

Related to https://github.com/wpilibsuite/allwpilib/issues/3872

calcmogul, 2022-01-07:
Here's a C++ test that reproduces the issue.
```cpp
#include <frc/TimedRobot.h>
#include <frc/simulation/SimHooks.h>
#include <gtest/gtest.h>
#include <hal/HAL.h>

class Robot : public frc::TimedRobot {};

TEST(Hang, Test) {
  frc::sim::PauseTiming();
  Robot robot;
  frc::sim::StepTiming(1_s);
  frc::sim::ResumeTiming();
}

int main(int argc, char** argv) {
    HAL_Initialize(500, 0);
    testing::InitGoogleTest(&argc, argv);
    return RUN_ALL_TESTS();
}
```

calcmogul, 2022-01-07:
Using StartCompetition() and EndCompetition() doesn't reproduce the issue.
```cpp
#include <thread>

#include <frc/TimedRobot.h>
#include <frc/simulation/SimHooks.h>
#include <gtest/gtest.h>
#include <hal/HAL.h>

class Robot : public frc::TimedRobot {};

TEST(Hang, Test) {
  frc::sim::PauseTiming();

  Robot robot;
  std::thread robotThread{[&] { robot.StartCompetition(); }};

  frc::sim::StepTiming(1_s);

  robot.EndCompetition();
  robotThread.join();

  frc::sim::ResumeTiming();
}

int main(int argc, char** argv) {
    HAL_Initialize(500, 0);
    testing::InitGoogleTest(&argc, argv);
    return RUN_ALL_TESTS();
}
```

virtuald, 2022-01-07:
I think that if a user calls startCompetition() part of the contract is that they must call endCompetition(). But if they didn't 'start' the object, I don't think it's sensible that it hangs.

calcmogul, 2022-01-07:
Not calling startCompetition() is violating the implicit contract of Notifier as well. It's expected that if you construct a Notifier, you will wait on it at some point. HALSIM_StepTiming() relies on this behavior when it calls WaitNotifiers() before stepping the timing. This ensures all Notifiers are in a known state.

WaitNotifiers() won't wait for a Notifier if the active flag is false. This only occurs when HAL_StopNotifier() is called. Not creating the Notifier in the TimedRobot constructor so the active flag is false would violate RAII.

I'm going to try initializing active to false instead of true, and have HAL_UpdateNotifierAlarm() set it to true.

Starlight220, 2022-01-09:
Is there a chance that this is what causes the WatchdogTest flakiness that we see in CI?

calcmogul, 2022-01-09:
The Watchdog test fail isn't a hang.