wpilibsuite/allwpilib #3891: WPILibNewCommands.json prevents performing online vendor dep updates

jcorcoran, 2022-01-10:
**Describe the bug**
A clear and concise description of what the bug is.

When attempting to perform online vendor library update check I receive the following error:

[image]

I believe the cause is the empty `"jsonUrl": "",` parameter in `WPILibNewCommands.json`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project from the java `Command Robot` template
2. Open the command pallet in VSCode, select `WPILib: Manage Vendor Libraries`
3. Select `Check for updates: Online`
4. Receive error in screenshot above.

This is a contrived simple demo, the error in this case is inconsequential (you can perform a WPIlib by other means), but this error prevents checking online for other 3rd party vendor deps which may be installed.

**Expected behavior**
A clear and concise description of what you expected to happen.

Checking online for 3rd party vendor deps should complete without error using the stock demo projects.

**Screenshots**
If applicable, add screenshots to help explain your problem.

See above.

**Desktop (please complete the following information):**
 - WPILib Version: v2022.1.1
 - OS: Windows 10
 - Java version openjdk version "11.0.13" 2021-10-19
 - C++ versio: N/A

**Additional context**
Add any other context about the problem here.

N/A

ThadHouse, 2022-01-10:
This is a known issue, that will be fixed in the next update. The pr and issue are in the vscode-wpilib repo.

closed 2022-01-10 by ThadHouse

jcorcoran, 2022-01-10:
👍 sorry looked for the bug before submitting, but didn't see anything ... looked in wrong place

for future travelers: https://github.com/wpilibsuite/vscode-wpilib/pull/515