wpilibsuite/allwpilib #3892: Kickoff release of WPILib appears to include log4j vulnerabilities

mjhealy, 2022-01-10:
**Describe the bug**
Based on feedback from the security tools on my laptop, the "WPILib 2022.1.1 Kickoff Release" appears to include several files that may contain vulnerable versions of "log4j".  The impacted files appear to be:
* wpilib/2022/maven/org/apache/logging/log4j/log4j-core/2.14.1/log4j-core-2.14.1.jar
* wpilib/2022/maven/gradle/plugin/com/github/jengelman/gradle/plugins/shadow/7.0.0/shadow-7.0.0.jar

**To Reproduce**
Steps to reproduce the behavior:
1. Run the "WPILib 2022.1.1 Kickoff Release" installer under MacOS.
2. Run security scans on the installed files.

**Expected behavior**
No problems should be reported with the installed files.

**Actual behavior**
Security tools report (in part):
> As part of the incident response for the ongoing Log4j Vulnerability Code Red, we have identified the following vulnerable program(s) on your machine: *<see list of files, above>*
>
> Due to the severity of this vulnerability, we need to identify and fix issues as quickly as possible. Therefore, we have automatically removed a Java class file from the above programs to ensure that an attacker can't exploit your machine.
>
> While we have tried to do this in a non-disruptive way, it is possible that the programs listed above may no longer function correctly. We are deeply sorry for any inconvenience.
>
> If you encounter issues after the reboot, please uninstall the affected program(s). If the developer of the program(s) has already released a patch to address the security issue, you may download and install the latest version. Otherwise, please wait until such a patch has been created before attempting to reinstall.

**Desktop (please complete the following information):**
 - WPILib Version: WPILib 2022.1.1 Kickoff Release
 - OS: MacOS 11.6.2

kasemir, 2022-01-10:
Taking the broader view of control system security, an FRC setup with known account passwords on the roborio, generally accessible network tables etc. should certainly not be used to control a power plant, maybe not even your garage door. The FRC control system is clearly meant to run on an isolated subnet like the robot's wifi in the lab or the field network during competitions.

mjhealy, 2022-01-10:
Ah.  That's good to know.  Thanks for the clarification!

PeterJohnson, 2022-01-16:
The shadow plugin was updated in 2022.2.1.

sciencewhiz, 2022-01-16:
Reopening as log4j 2.14.1 is still in the 2022.2.1 installer

calcmogul, 2022-01-16:
I assume this is because we forgot to merge https://github.com/wpilibsuite/WPILibInstaller-Avalonia/pull/235.

mjhealy, 2022-01-25:
The shadow-7.0.0.jar is also still in the installer.

Any word as to when another update with these items removed will be available?  (Still getting spanked by corp security tools because of their presence.)

calcmogul, 2022-01-25:
As I mentioned in the comment before yours, we forgot to merge https://github.com/wpilibsuite/WPILibInstaller-Avalonia/pull/235, which is why 2022.2.1 still has shadow 7.0.0. The next release will have shadow 7.1.2. This will happen as soon as NI releases a new FPGA (I have no idea when this will occur).

sciencewhiz, 2022-01-30:
> Any word as to when another update with these items removed will be available? (Still getting spanked by corp security tools because of their presence.)

The next update will have the new versions, but the installer will install them beside the old versions. As stated previously, they aren't used for developing robot code, so go ahead and delete them. You'd have to do that anyway when the new version is released.

sciencewhiz, 2022-02-08:
Fixed in WPILib 2022.3.1

closed 2022-02-08 by sciencewhiz