wpilibsuite/allwpilib #3895: VSCode Update Vendor Libraries Fails

Crossle86, 2022-01-10:
**Describe the bug**
Request to Manage Vendor Libraries, Check for Updates Online fails with message:
Command Manage Vendor Libraries resulted in an error (running contributed command: wpilibcore.manageVendorLibs failed)

**To Reproduce**
Steps to reproduce the behavior:
Select WPI  Extension command Manage Vendor  Libraries
Select Check for Updates Online

**Expected behavior**
A clear and concise description of what you expected to happen.
Vendor libraries should be  displayed with versions installed and version available online to be selected for update.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 WPILib 2022.1.1]
 - OS: [ Windows 7]
 - Java version [As installed by WPILib installer]

While there are other vendor library update issues, their descriptions did not match this so I added it as a new issue.

ThadHouse, 2022-01-10:
Hmm. What vendor libraries do you have installed. I wouldn't be surprised if Windows 7 didn't work with the workaround for lets encrypt certficiates.

Also, since I do see Windows 7, you did see the notice that Windows 7 support will be removed for next year, right? We already have plans that will be breaking support, and will not be implementing workarounds.

Crossle86, 2022-01-10:
I am aware of end of support for W7 and will transition...later. Too busy
right now. But I expect things to work as the W7 deprecation notice was
only officially announced with 2022 doc release.

The vendor
libraries are:
CTRE/Phoenix
NavX
RevColorSensorv3
Our
team library

Note that the command fails immediately with no
list of installed libraries.

On Mon, January 10,
2022 12:43 pm, Thad House wrote:
> Hmm. What vendor libraries do
you have installed. I wouldn't be surprised
> if Windows 7 didn't
work with the workaround for lets encrypt
> certficiates.
>
> Also, since I do see Windows 7, you did see the notice
that Windows 7
> support will be removed for next year, right? We
already have plans that
> will be breaking support, and will not
be implementing workarounds.
>
> --
> Reply to
this email directly or view it on GitHub:
>
https://github.com/wpilibsuite/allwpilib/issues/3895#issuecomment-1009325943
> You are receiving this because you authored the thread.
>
> Message ID:
***@***.***>

Richard Corn
Software Devices LLC
P.O.
Box 12299, Olympia WA. 98508
***@***.***Â Â Â www.swdevs.com

ThadHouse, 2022-01-10:
Oh, I think I know the issue. Does your custom lib also have an empty json url? I wouldn't be surprised if having 2 jsons missing urls would cause that message, and it won't know how to unify the exceptions and will just print a default one.

Crossle86, 2022-01-10:
I assume you are talking about the item "jsonUrl" in a vendordep jason
file. None of our vendordep files have blank jsonUrl fields.

> Oh, I think I know the issue. Does your custom lib also have an
empty json
> url? I wouldn't be surprised if having 2 jsons
missing urls would cause
> that message, and it won't know how to
unify the exceptions and will just
> print a default one.
GitHub:
https://github.com/wpilibsuite/allwpilib/issues/3895#issuecomment-1009342322
***@***.***>

Richard Corn
Software Devices LLC
P.O.
Box 12299, Olympia WA. 98508
***@***.***Â Â Â www.swdevs.com

PeterJohnson, 2022-01-16:
Fixed in 2022.2.1 release.

closed 2022-01-16 by PeterJohnson