wpilibsuite/allwpilib #3905: Unable to build a new java project due to gradlerio module jcenter download failure 502 error

dachaeon, 2022-01-12:
This is a Windows 10 install of the 2022 kickoff release and the initial build of the project with no code changes.

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'Test2022'.
> Could not resolve all files for configuration ':classpath'.
   > Could not resolve edu.wpi.first:GradleRIO:2022.1.1.
     Required by:
         project : > edu.wpi.first.GradleRIO:edu.wpi.first.GradleRIO.gradle.plugin:2022.1.1
      > Could not resolve edu.wpi.first:GradleRIO:2022.1.1.
         > Could not get resource 'https://plugins.gradle.org/m2/edu/wpi/first/GradleRIO/2022.1.1/GradleRIO-2022.1.1.module'.
            > Could not GET 'https://jcenter.bintray.com/edu/wpi/first/GradleRIO/2022.1.1/GradleRIO-2022.1.1.module'. Received status code 502 from server: Bad Gateway

ThadHouse, 2022-01-12:
Gradle is currently having issues. Assuming you used the installer, disconnect from the internet and it should build.

dachaeon, 2022-01-12:
Thank you! That got us past the issue.

closed 2022-01-12 by dachaeon