wpilibsuite/allwpilib #3912: StateSpaceSimTest is Missing State Space Sim Tests

modelmat, 2022-01-13:
**Describe the bug**
https://github.com/wpilibsuite/allwpilib/blob/main/wpilibc/src/test/native/cpp/simulation/StateSpaceSimTest.cpp only has the one test of flywheelSim. Add the rest.

**To Reproduce**
Steps to reproduce the behavior:
1. Watch intently at your screen and observe the existence of only one test
2. Existential dread

**Expected behavior**
There be tests.

**Screenshots**
No.

**Additional context**
My head still hurts.

modelmat, 2022-01-13:
It seems that:

- https://github.com/wpilibsuite/allwpilib/blob/main/wpilibc/src/test/native/cpp/simulation/SingleJointedArmSimTest.cpp
- https://github.com/wpilibsuite/allwpilib/blob/main/wpilibc/src/test/native/cpp/simulation/ElevatorSimTest.cpp

exist, so I think it would be better to rename StateSpaceSimTest to FlywheelSimTest

calcmogul, 2022-01-13:
Yep. We should fix that.