wpilibsuite/allwpilib #3914: NetworkTables Data Transmitting Delay

Ardasak, 2022-01-13:
**Describe the bug**
We are trying to send alliance, period and location datas to our vision-processing code. So we have to send the data from Java to Python. We have faced a issue that the data comes with 2 seconds of delay. Does anybody know why is this happening and how to fix it?

**To Reproduce**
Steps to reproduce the behavior:
1. We wrote a Python script to fetch the data from Java (It works properly when data goes from Python to Python.)
2. We also wrote a Java program which will send the datas to the Python side.
3. When we ran our code, we saw the data comes with 1-2 seconds of delay. 

**Expected behavior**
We are expecting less delay, 2 seconds is just too much...

**Screenshots**
[Java Code](https://ibb.co/gv5Ztp8)
[Python Code](https://ibb.co/sJWZP2g)

**Desktop (please complete the following information):**
 - WPILib Version: 2021.3.1
 - OS: Windows 10
 - Java version JDK 11
 - Python Version: 3.10.0
 - Python NetworkTables Version: 2021.0.0

**Additional context**
Thanks for your help.

PeterJohnson, 2022-01-13:
The default flush rate is 0.1 seconds, so this is definitely not expected behavior.  The screenshots don’t show the entirety of the Python loop; is there any sleep call in the loop or something else that would slow down the output?  If you don’t have anything like that, your code is pegging the CPU and spamming standard output with prints.  This can fill up the output buffer and result in delayed output.  Can you post your full code?

Ardasak, 2022-01-13:
https://ibb.co/JcSG5bg is our full python code and https://ibb.co/W3bsxgd this is the full function in java, the show_datas function only puts datas on the smart dashboard. There is no sleeping call. We use Nvidia Jetson Nano to run our code.

PeterJohnson, 2022-01-13:
Try adding a sleep to your Python loop to reduce the amount of data being printed to the console. Something like 10-20 ms?  Yes it will add a bit of latency for now, but that can be fixed later by switching to using value listeners.

Ardasak, 2022-01-13:
I'll try. Would you mind if I inform you tomorrow? The day is over where I live :D.

closed 2022-01-14 by Ardasak

egeakman, 2022-01-14:
NetworkTables rocks!