wpilibsuite/allwpilib #3921: WPILibNewCommands.json has version 2020.0.0

chauser, 2022-01-15:
I don't know if this is a bug or not. The version string in `WPILibNewCommands.json` is `2020.0.0` (confirmed in the github source). It greatly confused me when I was looking at Manage Vendor Libraries in VSCode as it seems highly likely that the current code is much newer than 2020. It doesn't cause any operational problem.

LudumDareDevelopment, 2022-01-16:
I don't think the new command framework has changed from 2020, so I think this is normal.

chauser, 2022-01-16:
The commit history for files suggests that it has changed (unless those changes going back over a year have not yet been released).

ThadHouse, 2022-01-16:
That version number is actually unimportant for commands. If you look at the individual artifacts, they have a wpilib version, which will force them to always match wpilib. The only time those files ever have to change is if the format of the artifacts change

chauser, 2022-01-16:
Yeah, the issue is only related to the human confusion that can arise from seeing it appear in "Manage Vendor Libraries" with a 2020 version.
Not a big deal.

brhea, 2022-01-16:
I've been a bit confused by this myself and have spent some time in researching to verify that I'm using the proper version of the library..... easily 30 minutes to an hour.  Multiply that by the number of mentors in FRC and that's a lot of hours spent spinning wheels because of an inconsistency in the way rev levels are displayed.

PR #3938 (2022-01-21): Update command vendordep JSON version number
While the number doesn't matter, it being old is confusing a lot of
people.

Closes #3921.
commit 841174f3 (2022-01-21): [commands] Change command vendordep JSON version number to 1.0.0 (#3938)

While the number doesn't matter, it being old is confusing a lot of
people.  We never increment the internal vendordep versions, so using a year
version number was a poor choice.

Closes #3921.
closed 2022-01-21 by PeterJohnson