wpilibsuite/allwpilib #3924: Solenoid.startpulse() does not work with 2022 API

brhea, 2022-01-16:
After upgrading to 2022 wpilib and importing code from previous year, we found that none of our solenoids worked. We did put in the additional parameter in Solenoid to specify the pneumatic module type due to addition of REV pneumatic hub and it's likely that our problems resulted from the addition of the new PCM type.

Solenoid(...).startpulse() no longer works at all. 

We created a separate project with only code to manipulate the solenoid to try to isolate from other problems and it still did not work. 

Solenoid(...).set(true/false) method works. Solenoid will turn on and off as demanded.

We wrote a workaround to add our own pulse to the output using the set() method so this is not urgent for us but it is an issue to consider fixing.

sciencewhiz, 2022-01-16:
It isn't entirely clear from your report. Are you seeing this with a PCM or a PH?

brhea, 2022-01-16:
We're using the PCM from CTRE. We selected the CTRE module type and the CAN address was correct.

sciencewhiz, 2022-01-17:
It looks like the PCM one shot tests have been failing for a while

brhea, 2022-01-17:
This worked in our 2020 code. We did not update in 2021. The error showed up for us this year when we updated to 2022 wpilib.