wpilibsuite/allwpilib #3930: MecanumDrive class incorrectly assumes +Y is forward

calcmogul, 2022-01-17:
According to the documented NED axes convention, +X should be forward, so xSpeed == 1 in MecanumDrive.driveCartesian() should make alll the motors move forward. However, the existing implementation results in ySpeed == 1 making all the motors move forward. The implementation is wrong. The tests verify the existing behavior, so the tests are also wrong.

This class's behavior was originally compared to RobotDrive in tests, and RobotDrive uses +Y as forward. Therefore, it's likely MecanumDrive has been broken since it was written in 2016.

Note the following code:
```java
  public static WheelSpeeds driveCartesianIK(
      double ySpeed, double xSpeed, double zRotation, double gyroAngle) {
    ySpeed = MathUtil.clamp(ySpeed, -1.0, 1.0);
    xSpeed = MathUtil.clamp(xSpeed, -1.0, 1.0);

    // Compensate for gyro angle.
    Vector2d input = new Vector2d(ySpeed, xSpeed);
    input.rotate(-gyroAngle);

    double[] wheelSpeeds = new double[4];
    wheelSpeeds[MotorType.kFrontLeft.value] = input.x + input.y + zRotation;
    wheelSpeeds[MotorType.kFrontRight.value] = input.x - input.y - zRotation;
    wheelSpeeds[MotorType.kRearLeft.value] = input.x - input.y + zRotation;
    wheelSpeeds[MotorType.kRearRight.value] = input.x + input.y - zRotation;
```
The input vector taking ySpeed as the x-component is intentional, because Vector2d assumes NWU convention and the drive class assumes NED. However, the wheelSpeeds assignments don't take this into account when extracting the coordinates; `.x` and `.y` should be swapped. The broken tests reinforced this incorrect behavior.

We may want to look into taking NED convention on the API boundary, then converting to NWU for all internal calculations to avoid mistakes like this.

calcmogul, 2022-01-24:
Depends on whether teams rely on the current (broken) behavior or not. If they got the argument order backwards by accident, swapping the argument order in the API would fix it.

We opted to not silently change the function API or behavior mid-season.

rthinman, 2022-01-24:
I can't speak for anyone else.  We got this test bot working in 2020 by following the documented behavior of the function parameters, not thinking about NED/NWU, etc.  "OK, the first parameter is called 'y', that's weird, but the doc says the bot goes __ when it is positive.  The joystick gives __ on the __ axis when pushed that direction.  Give it that axis (perhaps inverted).  [repeat]."

Agreed on not changing API or behavior mid-season.  Until the PR above gets released, there could be confused people.  Maybe there aren't very many using mecanum; we certainly tend toward keeping it simple for competitions.

Starlight220, 2022-10-30:
Now that #4079 was merged, what's the status of this?

closed 2022-10-31 by calcmogul