wpilibsuite/allwpilib #3931: C++ library doesn't build on linux

Scoder12, 2022-01-18:
**Describe the bug**
C++ library does not build due to errors.

**To Reproduce**
On linux machine run `./gradlew build` as shown in the README.

**Expected behavior**
Successful build

**Screenshots**
```
error: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Werror=cpp]
  397 | #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
      |    ^~~~~~~
cc1plus: all warnings being treated as errors
```

**Desktop (please complete the following information):**
 - WPILib Version: 2022.1.1
 - OS: Tested on NixOS unstable and Ubuntu
 - Java version `openjdk 11.0.12 2021-07-20`
 - C++ version `g++ (GCC) 10.3.0`

**Additional context**
It would be nice if the build instructions were a bunch of bullet points concisely describing what each of the gradle commands currently described do.

calcmogul, 2022-01-18:
Looks like your OS's default CFLAGS or CXXFLAGS are interfering.

calcmogul, 2022-01-18:
> It would be nice if the build instructions were a bunch of bullet points concisely describing what each of the gradle commands currently described do.

How would you like https://github.com/wpilibsuite/allwpilib/blob/main/README.md#building to be modified in particular?

Scoder12, 2022-01-18:
> Looks like your OS's default CFLAGS or CXXFLAGS are interfering, or you have an alias of g++ that adds flags. We don't add -D_FORTIFY_SOURCE to our builds.

Not sure how I would fix that:
```
❯ env | grep -i flags
❯ command -v g++
/home/spencer/.nix-profile/bin/g++
```

Scoder12, 2022-01-18:
> > It would be nice if the build instructions were a bunch of bullet points concisely describing what each of the gradle commands currently described do.
> 
> How would you like [`main`/README.md#building](https://github.com/wpilibsuite/allwpilib/blob/main/README.md?rgh-link-date=2022-01-18T00%3A57%3A32Z#building) to be modified in particular? Is there information missing that you need?

Information is perfectly fine, just thought it could be a little more concise but it isn't a big deal at all.

sciencewhiz, 2022-01-18:
> Not sure how I would fix that:

It appears that you aren't the only one to have NixOS add flags that cause issues. You may be able to try some of the workarounds in these issues: https://github.com/NixOS/nixpkgs/issues/60919 https://github.com/NixOS/nixpkgs/issues/18995

What version of Ubuntu? It compiles on 20.04 and 18.04 for me.

PeterJohnson, 2022-01-18:
From the second issue, it looks like `export hardeningDisable=all` in your shell might be the workaround?

Scoder12, 2022-01-18:
Indeed, using a shell.nix with 
```nix
{ pkgs ? import <nixpkgs> { } }:
pkgs.mkShell {
  nativeBuildInputs = [ pkgs.openjdk11 pkgs.gcc ];
  hardeningDisable = [ "all" ];
}
```
did fix the issue. Just tried again on Ubuntu and it works there as well - I must have misremembered. 

Had no idea this would be a Nix specific issue, thanks for the help!

closed 2022-01-18 by Scoder12