wpilibsuite/allwpilib #3940: REV PDH and PCH stop responding on CAN bus

eohara0920, 2022-01-19:
**Describe the bug**
After using the code for some amount of time, the REV Pneumatic Hub and Power Distribution Hub stop sending data over the CAN bus and therefore the DoubleSolenoid set() function can no longer control Pneumatics

**To Reproduce**
Steps to reproduce the behavior:
Using REV Control System

In subsystem class create    

`private final DoubleSolenoid m_actuator = new DoubleSolenoid(2, PneumaticsModuleType.REVPH, 0, 1);`

create functions to use 
```
    public void extend(){
        m_actuator.set(Value.kForward);
    }
    public void retract(){
        m_actuator.set(Value.kReverse);
    }
```

call these functions from a Command

After some time the PDH and PCH will not respond to CAN bus unless power is cycled. They are on 2022 latest firmware. Utilizing a type C cable into a SparkMax controller does not show the PDH or PCH on the Rev Hardware Client either.

**Expected behavior**
PDH and PCH should continue to communicate on CAN bus

**Desktop (please complete the following information):**
 - WPILib Version: 2022.2.1

**Additional context**
I think this might actually be a REV specific firmware issue as the devices stop responding to CAN until power cycled, but thought I would post here as well.

eohara0920, 2022-01-19:
Stack Trace

[image]

sciencewhiz, 2022-01-19:
What is the order of magnitude of "some amount of time"?

eohara0920, 2022-01-20:
Sometimes less than 30 seconds sometimes a few minutes. We disconnected the power distribution hub and it seemed to help

PeterJohnson, 2022-01-20:
@NoahAndrews

NoahAndrews, 2022-01-20:
@eohara0920 also submitted a ticket to our support system, and we're going to gather additional information there. We'll update this thread once we have information to share.

NoahAndrews, 2022-01-27:
This issue has been fixed in PDH firmware 22.0.2 and PH firmware 22.0.2. You can update using the REV Hardware Client.

Daltz333, 2022-01-28:
Closing as fixed.

closed 2022-01-28 by Daltz333