wpilibsuite/allwpilib #3941: Addressable LEDs duplicate each LED

jasondaming, 2022-01-20:
When setting AddressableLEDs using WPILib the buffer you set will have each of the first half of the entries repeated.  For example if you set [green, purple, blue, red] it will give you [green, green, purple, purple].  I suspect this might be at a lower level than WPILib but until the cause is known I think we should document it here.

Code used to reproduce:
```
private final AddressableLED m_led = new AddressableLED(LEDConstants.kLEDPWMPort);
  private final AddressableLEDBuffer m_ledBuffer = new AddressableLEDBuffer(4);

  /** The container for the robot. Contains subsystems, OI devices, and commands. */
  public RobotContainer() {
    // Configure the button bindings
    configureButtonBindings();
    m_ledBuffer.setLED(0, Color.kGreen);
    m_ledBuffer.setLED(1, Color.kPurple);
    m_ledBuffer.setLED(2, Color.kBlue);
    m_ledBuffer.setLED(3, Color.kRed);
    m_led.setLength(m_ledBuffer.getLength());
    m_led.setData(m_ledBuffer);
    m_led.start();
  }
  ```

mjhealy, 2022-01-22:
Thanks for logging this, Jason: we ran into it while working with some test code today, and I was just about to log a bug report when I saw yours. :-)

closed 2022-02-01 by PeterJohnson

mjhealy, 2022-02-01:
This is marked as closed, but presumably won't land for teams until an upcoming release of WPILib.  #3944 makes it look like one is likely to arrive soon: any guesses on when we should be prepping for it?

calcmogul, 2022-02-01:
No idea. It depends on when NI releases new installers.

PR #3944 (2022-02-01): Update to 2022 v4.0 image
Closes #3941