wpilibsuite/allwpilib #3945: Eigen::Vector is not a member of 'Eigen'

marcomasa, 2022-01-20:
**Describe the bug**

I am trying to use the trajectory generation tools in a ROS C++ application.
I successfully installed the lib and can include and use some tools, but not the ones i actually need.
When i am building my package, i get the following error:

´´´
/usr/local/wpilib/include/wpimath/frc/spline/Spline.h: In member function ‘frc::Spline<Degree>::PoseWithCurvature frc::Spline<Degree>::GetPoint(double) const’:
/usr/local/wpilib/include/wpimath/frc/spline/Spline.h:58:12: error: ‘Vector’ is not a member of ‘Eigen’; did you mean ‘VectorXd’?
   58 |     Eigen::Vector<double, Degree + 1> polynomialBases;
      |            ^~~~~~
´´´

**To Reproduce**
Steps to reproduce the behavior:
1. Download allwpilib-2022.2.1
2. Follow CMake Install instructions (Options below)
3. Create own ROS package and #include "frc/trajectory/TrajectoryConfig.h"
4. Build package

OPTIONS
´´´
# Options for building certain parts of the repo. Everything is built by default.
option(BUILD_SHARED_LIBS "Build with shared libs (needed for JNI)" ON)
option(WITH_JAVA "Include java and JNI in the build" OFF)
option(WITH_CSCORE "Build cscore (needs OpenCV)" OFF)
option(WITH_WPIMATH "Build wpimath" ON)
option(WITH_WPILIB "Build hal, wpilibc/j, and myRobot (needs OpenCV)" ON)
option(WITH_OLD_COMMANDS "Build old commands" OFF)
option(WITH_EXAMPLES "Build examples" OFF)
option(WITH_TESTS "Build unit tests (requires internet connection)" OFF)
option(WITH_GUI "Build GUI items" OFF)
option(WITH_SIMULATION_MODULES "Build simulation modules" OFF)
´´´

Relevant CmakeList Options for custom package:
´´´
cmake_minimum_required(VERSION 3.0.2)
project(my_package)
add_compile_options(-std=c++17)
find_package(wpilib REQUIRED)
...
 target_link_libraries(my_package
   ${catkin_LIBRARIES}
   wpimath
 )
´´´

**Expected behavior**
Clean build

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: 2022.2.1
 - OS: Linux Ubuntu 20.04
 - C++ version [e.g. 17]

**Additional context**
- Using ROS Noetic
- Using catkin packages:

´´´
find_package(catkin REQUIRED COMPONENTS
  costmap_2d
  evo_nav_database
  geometry_msgs
  nav_core
  nav_msgs
  roscpp
  rospy
  tf
  pluginlib
  global_planner
  pcl_ros
)
´´´

marcomasa, 2022-01-20:
The bug does not appear when using v2021.3.1

However, i needed to change something in the CMakeList Options:
```
find_package(wpilib REQUIRED)
-- changes to --->
find_package(wpimath REQUIRED)
```

marcomasa, 2022-01-20:
Using the above solution does not work for 2022.2.1

calcmogul, 2022-01-20:
Eigen::Vector is a typedef added in Eigen 3.4. What version do you have installed?

marcomasa, 2022-01-20:
I have version 3.3.7-2.
Trying to update via apt is not working, it tells me i already have the newest version.
(Linux Ubuntu 20.04.2 LTS)

PeterJohnson, 2022-01-20:
Sounds like our cmake build isn’t installing/pointing to our embedded copy of Eigen?

calcmogul, 2022-01-20:
The CMake build works fine for me without Eigen installed. I wonder if the following lines prioritize a system install.
```cmake
    find_package(Eigen3 CONFIG REQUIRED)
    target_link_libraries (wpimath Eigen3::Eigen)
```

PeterJohnson, 2022-01-20:
We shouldn’t be looking for a system install at all since we embed it.

PeterJohnson, 2022-01-21:
Those lines are only used if USE_VCPKG_EIGEN is defined.

calcmogul, 2022-01-21:
Then idk why it would be picking up an Eigen other than ours.

PeterJohnson, 2022-01-21:
I agree.  @marcomasa do you see a wpimath/Eigen/Core file in the installation directory?  Can you show what the compiler command line is so we can check the -I options?

marcomasa, 2022-01-23:
@PeterJohnson  yes i have some Eigen files:

```
-- Installing: /usr/local/wpilib/include/wpimath
-- Installing: /usr/local/wpilib/include/wpimath/unsupported
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/MatrixFunctions
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions/MatrixSquareRoot.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions/StemFunction.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions/MatrixFunction.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions/MatrixPower.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions/MatrixExponential.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/MatrixFunctions/MatrixLogarithm.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/AutoDiff
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/AutoDiff/AutoDiffScalar.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/AutoDiff/AutoDiffVector.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/src/AutoDiff/AutoDiffJacobian.h
-- Installing: /usr/local/wpilib/include/wpimath/unsupported/Eigen/AutoDiff
-- Installing: /usr/local/wpilib/include/wpimath/Eigen
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/Eigenvalues
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/Jacobi
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/LU
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/StdDeque
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/QR
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/Householder
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/StdList
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/SVD
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/SelfAdjointEigenSolver.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/HessenbergDecomposition.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/ComplexEigenSolver.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/EigenSolver.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/GeneralizedEigenSolver.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/RealQZ.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/ComplexSchur.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/RealSchur.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/MatrixBaseEigenvalues.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/Tridiagonalization.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Eigenvalues/GeneralizedSelfAdjointEigenSolver.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/StlSupport
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/StlSupport/StdList.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/StlSupport/StdDeque.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/StlSupport/details.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/StlSupport/StdVector.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Jacobi
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Jacobi/Jacobi.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU/FullPivLU.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU/InverseImpl.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU/PartialPivLU.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU/arch
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU/arch/InverseSize4.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/LU/Determinant.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/misc
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/misc/Kernel.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/misc/Image.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/misc/RealSvd2x2.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/ArrayCwiseUnaryOps.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/ReshapedMethods.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/CommonCwiseUnaryOps.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/MatrixCwiseUnaryOps.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/MatrixCwiseBinaryOps.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/IndexedViewMethods.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/ArrayCwiseBinaryOps.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/BlockMethods.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/plugins/CommonCwiseBinaryOps.h
[... 132 lines cut ...]
[... cut ...]
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Core/CoreIterators.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Core/Ref.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/src/Core/CoreEvaluators.h
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/Cholesky
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/StdVector
-- Installing: /usr/local/wpilib/include/wpimath/Eigen/Core
-- Installing: /usr/local/wpilib/lib/libwpimath.so
-- Set runtime path of "/usr/local/wpilib/lib/libwpimath.so" to "/usr/local/wpilib/lib"
```

Starlight220, 2022-08-29:
Is there an actionable item from here or can this be closed?

closed 2022-08-29 by calcmogul