wpilibsuite/allwpilib #3953: ShuffleboardTab can't return the same widget twice.

chenPerach, 2022-01-22:
example code:
```Java
public class Robot extends TimedRobot {
  ShuffleboardTab tab;
  @Override
  public void robotInit() {
    tab = Shuffleboard.getTab("test");
  }
  @Override
  public void robotPeriodic() {
    NetworkTableEntry e = tab.add("t",0).getEntry();
  }
}
```

this code will crash because the user is requesting the same widget from the ```"test"``` tab.

**error**:
```
Unhandled exception: java.lang.IllegalArgumentException: Title is already in use: t
```

## the reason for my confusion

instead of throwing an exception, why won't the shuffleboardTab just return the requested widget?
from what I have seen the tab checks if the title already exists in a string set called ```m_usedTitles```
and if it does it throws an ```IllegalArgumentException```.

instead of that why don't we use a map containing all of the already existing widgets.
and return the widget to the user if it already exists in that map.

### the corresponding code related to the question
ContainerHelper.java
```Java
  private void checkTitle(String title) {
    if (m_usedTitles.contains(title)) {
      throw new IllegalArgumentException("Title is already in use: " + title);
    }
    m_usedTitles.add(title);
  }
```

Daltz333, 2022-01-22:
Unfortunately, Shuffleboard is basically in a extended support stage. What this means is that non-breaking bugs or features aren't likely to get fixed/implemented without external contributors. 

If you are interested in contributing, feel free to open a PR. I'm going to tag this as help wanted.

PeterJohnson, 2022-01-22:
It’s an intentional design decision.  All of the add() functions should only be called once for a particular widget.  The expectation is that all of the setup is done in robotInit().

chenPerach, 2022-01-22:
> It’s an intentional design decision. All of the add() functions should only be called once for a particular widget. The expectation is that all of the setup is done in robotInit(). Is there a particular reason you want to call it multiple times / call it from periodic?

yes, we want to create different tabs for each subsystem and command we are currently debugging.
this makes pid calibration and bug fixing much easier. 

to do that we had to write classes to work around this exception that seemed unintuitive and weird.
we wonderd what was the reason for this type of implementation.

ThadHouse, 2022-01-25:
Its actually pretty standard Java convention. When you call "add" on a named collection, if that element already exists, an exception will be thrown rather then returning the existing item. Normally a different API is used if you want to override if it already exists, or to get the existing item. You could also store either the originally created item or the network table entry directly.

Starlight220, 2022-11-22:
If we don't intend to change this behavior, we should at least document it.

Starlight220, 2022-11-22:
On second look, it appears that in C++ the existing tab instance is returned and no exception is thrown.

Starlight220, 2023-12-22:
On third look, the only difference between Java and C++ is that Java throws an exception while C++ only reports an error.

closed 2023-12-22 by Starlight220