wpilibsuite/allwpilib #3956: Unable to run simulator "cscorejni could not be loaded"

allengregoryiv, 2022-01-23:
I have a PC that I can't get to run the simulator, I have installed the C++ runtime from the link, the Visual Studio Desktop C++ tools, installed WPILib for all users and just myself, ran WPIlib as admin, restarted the PC many times after installing everything but I still can't get it to work. 

Mostly just wondering if there is anything else I should try, I'm hoping I'm just missing something.

Windows 10, school lab PC, so not sure if IT set up the image in a strange way and there is something else that needs to be installed or run. I have admin access to this machine.

"java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
        attempted to load for platform /windows/x86-64/
Last Load Error:
C:\github\GammaRay-2021-2022\build\jni\release\cscorejni.dll: Can't find dependent libraries
A common cause of this error is missing the C++ runtime.
Download the latest at https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads

        at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
        at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:430)
        at frc.robot.Main.main(Main.java:20)"

ThadHouse, 2022-01-23:
Is the system running Windows 10 N edition? The N editions are often installed by IT departments, and don't have the media libraries needed by cscore. You can install the media pack to add those libraries back.
https://www.microsoft.com/en-us/software-download/mediafeaturepack

allengregoryiv, 2022-01-23:
Thank You Thad, that did it.

The media pack installer didn't work for my version of windows and the instructions to add features through settings didn't work either. But the last post here has the instructions to use DISM to install the media pack and that solved the issue for anyone else that's in this situation in the future. - https://www.tenforums.com/software-apps/170733-installing-windows-media-feature-pack.html

ThadHouse, 2022-01-25:
Since this is documented in the known issues in the docs, I'm going to close this.

closed 2022-01-25 by ThadHouse