wpilibsuite/allwpilib #3957: Creating 2 ADXRS450_Gyro devices in simulation hangs the robot

virtuald, 2022-01-24:
**Describe the bug**
Creating 2 ADXRS450_Gyro devices in simulation hangs the robot program.

**To Reproduce**

```
$ python
Python 3.10.1 (main, Dec  7 2021, 00:00:00) [GCC 11.2.1 20210728 (Red Hat 11.2.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wpilib
>>> wpilib.ADXRS450_Gyro()
<wpilib._wpilib.ADXRS450_Gyro object at 0x7fbf28f091b0>
>>> wpilib.ADXRS450_Gyro()
^C^C^C^C^Z
[1]+  Stopped                 python
```

**Expected behavior**
An exception saying "device already exists" or something to that effect.

**Additional context**

Obviously this falls under the category of "Well don't do that"... but I wonder whether there are other difficult to diagnose bugs in places where the simulation behavior hinges on whether (bool)simDevice is true?

ThadHouse, 2022-01-25:
Its actually more invasive to do this mid season then I would like, so I think this will have to wait until 2023.

For future me, its because the low level does not use handles, so it doesn't have great checking. And then at the high level we only report an error on failure and not throw an initialization exception.

PeterJohnson, 2022-01-25:
We likely have several classes affected by this, essentially what's happening is we use the simDevice return value as a proxy for is-simulation.  A relatively low impact change would be to check the false case of that (e.g. failure to create a SimDevice) and throw an error only if simulation is enabled (which would indicate a double-allocation in simulation).

Starlight220, 2022-11-27:
I wasn't able to reproduce the hang in Java/C++.

In Java, we can check and throw in `SimDevice.create`, but in C++ `SimDevice` doesn't have access to `frc/Errors.h`.

virtuald, 2022-11-30:
This still occurs on the current RobotPy beta.

PeterJohnson, 2025-02-07:
This class was removed for 2027.