wpilibsuite/allwpilib #3963: Sim gui does not import 2022 field json

Crossle86, 2022-01-26:
The sim gui that appears when you run sim in VSC has a problem with the Field widget. I can choose a field image json file for 2021 field the field is imported and displayed. When I try to import the 2022-rapidreact.json file, nothing happens. Previously loaded image remains displayed. If I import the 2022-field.png it displays the 2022 field image. 

**Desktop (please complete the following information):**
 - WPILib Version: [e.g. 2022.2.1]
 - OS: [e.g. Windows 7]
 - Java version [per 2022 WPILib installer]

PeterJohnson, 2022-01-27:
I can't reproduce this.  My setup (Linux): clone the PathWeaver repo (https://github.com/wpilibsuite/PathWeaver/tree/main/src/main/resources/edu/wpi/first/pathweaver) and load the 2022 json file from that location, and Sim GUI displays it without an issue.

Crossle86, 2022-01-27:
My apologies. I downloaded the files separately, not in the zip. Apparently I made some error doing that as the files out of the repo zip work as expected.

closed 2022-01-27 by Crossle86