wpilibsuite/allwpilib #3978: Sysid Generates Broken Code for TalonFX devices on 2022.2.1 update

lospugs, 2022-02-01:
**Describe the bug**
When using Sysid to deploy a project, the resulting robot code that is deployed generates code that is not functional 

**To Reproduce**
Steps to reproduce the behavior:
1. Open Sysid
2. Configure a Drivetrain characterization with TalonFX
3. Set L motor to CAN ID 1, Set R motor to inverted and CAN ID 2
4. Select Navx as the Gyro on the SPI.kMXP port
5. Set Encoder to 2048 
6. Set gear ratio to 10.71:1
7. Deploy
8. Run the tests
8. Enable Autonomous Mode, and view either RioLog or DriverStation console

**Expected behavior**
I expected the robot to start accelerating slowly, travel a short distance, and stop when I disable it.

Behavior is identical from MacOS and Windows 10, recently updated (29 January).

**Desktop (please complete the following information):**
 - WPILib Version: 2022.2.1
 - OS: MacOS Monterey, and Windows 10 x64
 - Java version jdk11
 - C++ version unknown

**Additional context**
Here is a dump of the console while running:
`Killing previously running FRC program...
FRC pid 4216 did not die within 500 ms. Force killing with kill -9
********** Robot program starting **********
Camera Server Library Not Found
Initializing logger
Read JSON
Setup motors
NT: server: client CONNECTED: 10.18.95.49 port 55120
CTR: No new response to update signal
CTR: No new response to update signal
Setup TalonFXSetup TalonFXSetup encoders
Setup Built-in+TalonFXSetup Built-in+TalonFXSetup gyro
Setup NavX, Port: SPI.kMXPInstantiating navX-Sensor on SPI Port 4.
********** Robot program startup complete **********
Robot disabled
Collected: 0 data points.
navX-Sensor startup initialization and startup calibration complete.
navX-Sensor onboard startup calibration complete.
navX-Sensor Yaw angle auto-reset to 0.0 due to startup calibration.
Warning: Watchdog not fed within 0.005000s
Warning at Main: Warning: Watchdog not fed within 0.005000s
at frc::Watchdog::Impl::Main() + 0x1a8 [0xb6c96218]
at + 0x9d284 [0xb64ce284]
Warning: LiveWindow::UpdateValues(): 0.000008s
RobotPeriodic(): 0.014026s
SmartDashboard::UpdateValues(): 0.000056s
DisabledPeriodic(): 0.000043s
Shuffleboard::Update(): 0.000008s
Warning at PrintEpochs: Warning: LiveWindow::UpdateValues(): 0.000008s
RobotPeriodic(): 0.014026s
SmartDashboard::UpdateValues(): 0.000056s
DisabledPeriodic(): 0.000043s
Shuffleboard::Update(): 0.000008s
at frc::Tracer::PrintEpochs() + 0x124 [0xb6c89808]
at frc::IterativeRobotBase::LoopFunc() + 0x150 [0xb6c43cf0]
at frc::TimedRobot::StartCompetition() + 0x444 [0xb6c13c14]
at void frc::impl::RunRobot<Robot>(wpi::priority_mutex&, Robot**) + 0x54 [0x2ac74]
at int frc::StartRobot<Robot>() + 0x3d4 [0x2b1b4]
at __libc_start_main + 0x114 [0xb6239580]
Error: Loop time of 0.005000s overrun
Error at PrintLoopOverrunMessage: Error: Loop time of 0.005000s overrun
at frc::IterativeRobotBase::PrintLoopOverrunMessage() + 0x74 [0xb6c44190]
at frc::Watchdog::Impl::Main() + 0x90 [0xb6c96100]
at + 0x9d284 [0xb64ce284]
[phoenix-diagnostics] Server 1.9.0 (Dec 21 2021,02:10:43) running on port: 1250
CTR: CAN frame not received/too-stale.
[phoenix] Library initialization is complete.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.
CTR: CAN frame not received/too-stale.`

Daltz333, 2022-02-01:
Hm. Those errors lead me to believe this is an electrical problem. Can you verify that a robot project you program yourself works as intended?

lospugs, 2022-02-01:
Yes, [our code works just fine ](https://github.com/Lambda-Corps/2022_RapidReact/blob/main/src/main/java/frc/robot/subsystems/DriveTrain.java)without any of these messages. Phoenix Tuner can properly drive the TalonFX by CAN ID without any error messages.

The SysID tool characterization worked fine with this exact robot as wired on the 28th of January (Friday) but not after we updated to 2022.2.1 on the 29th (Saturday).