wpilibsuite/allwpilib #3984: CameraServer: Expose number of clients

agasser, 2022-02-02:
It would be useful if CameraServer exposed the number of clients connected to the server for streaming video.

In a Python image pipeline, there may be additional work to prepare each frame for streaming. For example, the image may need to be resized and have its color converted from RGB to BGR. If there are no clients for the CameraServer this additional processing can be skipped, potentially increasing FPS for the pipeline.

Streaming the video from the imaging pipeline would often be something useful in the shop for debugging, but not useful during a competition, when FPS matters most.

There are workarounds, such as using a NetworkTable entry to toggle streaming.

This CD post has additional detail: [CameraServer - is anyone watching?](https://www.chiefdelphi.com/t/cameraserver-is-anyone-watching/402008)

sciencewhiz, 2022-02-02:
Do you actually need count or just if anyone is connected. See csSource.isEnabled()

https://www.chiefdelphi.com/t/cameraserver-is-anyone-watching/402008/6

agasser, 2022-02-02:
I think `csSource.isEnabled()` would do the trick. It's missing from robotpy so I didn't see it. I'll created an issue over there.
https://github.com/robotpy/robotpy-cscore/issues/111

closed 2022-02-02 by agasser

PeterJohnson, 2022-02-02:
As noted in that thread, while the exact number of clients is not exposed, VideoSource::IsEnabled() can be used to determine on the source side if there are 1 or more active clients.