wpilibsuite/allwpilib #3998: How do I utilize a package from a custom Maven repo?

EddieDL, 2022-02-05:
**Describe the question you have.**
How do I utilize a package from a custom Maven repo?

**Describe the reason for your confusion.**
**Is your question related to a problem? Please describe.**
I am trying to utilize a package from a custom Maven repo. I have configured this by adding the following to my settings.gradle:
```
maven { url "https://maven.brott.dev/" }
```
However, whenever I build the project it can't find the dependency and when it says where it checked it only checked local and https://repo.maven.apache.org/maven2

ThadHouse, 2022-02-05:
The maven definitions in settings.gradle are only for build plugins. For actual code, you need to put that in a repositories block in build.gradle.

EddieDL, 2022-02-05:
Thank you, that was just what I needed.

closed 2022-02-05 by PeterJohnson

sciencewhiz, 2022-02-05:
https://docs.wpilib.org/en/stable/docs/software/advanced-gradlerio/external-libraries.html