wpilibsuite/allwpilib #4002: Robot code crashes when initializing a REV PH after a CAN Spark Max and no bus attached

willtoth, 2022-02-06:
**Describe the bug**
When running a roboRIO or roboRIO 2 with the can bus disconnected, and a DoubleSolenoid with type REVPH is created after a CANSparkMax, the robot crashes.

**To Reproduce**
Example code added to TimedRobot template

```java
  private CANSparkMax m_sparkMax = new CANSparkMax(1, MotorType.kBrushless);
  private DoubleSolenoid m_solenoid = new DoubleSolenoid(
    PneumaticsModuleType.REVPH, 0, 1);
```

Running this code crashes the robot. However, flipping this order, or removing either initialization does not crash the robot.

Here is an example stack trace

```
NT: server: client CONNECTED: 172.22.11.1 port 63090 ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Unhandled exception instantiating robot edu.wpi.first.hal.REVPHJNI edu.wpi.first.hal.util.UncleanStatusException:  Code: -35007. HAL: CAN Output Buffer Full. Ensure a device is attached ﻿﻿ frc.robot.Robot.<init>(Robot.java:31) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at frc.robot.Robot.<init>(Robot.java:31): Unhandled exception instantiating robot edu.wpi.first.hal.REVPHJNI edu.wpi.first.hal.util.UncleanStatusException:  Code: -35007. HAL: CAN Output Buffer Full. Ensure a device is attached ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.hal.REVPHJNI.initialize(Native Method) ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ The robot program quit unexpectedly. This is usually due to a code error.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information.
 ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:336) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.PneumaticHub$DataStore.<init>(PneumaticHub.java:32) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.PneumaticHub.getForModule(PneumaticHub.java:103) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.PneumaticHub.<init>(PneumaticHub.java:142) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.PneumaticsBase.getForType(PneumaticsBase.java:19) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.DoubleSolenoid.<init>(DoubleSolenoid.java:62) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.DoubleSolenoid.<init>(DoubleSolenoid.java:45) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.Robot.<init>(Robot.java:31) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:322) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:463) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.Main.main(Main.java:23) ﻿
﻿﻿﻿﻿﻿﻿  ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Could not instantiate robot edu.wpi.first.hal.REVPHJNI! ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:342) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:336): The robot program quit unexpectedly. This is usually due to a code error. ﻿
﻿﻿﻿﻿﻿﻿   The above stacktrace can help determine where the error occurred. ﻿
﻿﻿﻿﻿﻿﻿   See https://wpilib.org/stacktrace for more information. ﻿
﻿﻿﻿﻿﻿﻿  ﻿
﻿﻿﻿﻿﻿﻿ Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:342): Could not instantiate robot edu.wpi.first.hal.REVPHJNI! ﻿

```

**Expected behavior**
The robot controller does not crash, instead an error is shown.

**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1
 - OS: Windows 10
 - Java version: 11

**Additional Information**
The above stack trace is with an example project. It is similar with team code but without `ERROR ﻿﻿ 1 ﻿﻿ Could not instantiate robot edu.wpi.first.hal.REVPHJNI!`

Starlight220, 2022-10-31:
Was this closed by #4169?

PeterJohnson, 2022-10-31:
Yes

closed 2022-10-31 by PeterJohnson