wpilibsuite/allwpilib #4008: Strange behavior from wpi::Now()

jlmcmchl, 2022-02-06:
wpi::Now() appears to sometimes return nanosecond- precision timestamps, or is otherwise showing strange behavior. Perhaps the clock is speeding up after DS disconnect? I'm not sure what's going on, but it doesn't seem right. 

We've been testing out the DataLogger that Peter is building - I would expect a relatively straight line, showing that we're logging data at the same rate across the session. However, time seems to speed up after 15 or so seconds:
[image]

**Does this appear out of the ordinary?**

**Steps to Reproduce:**
I don't have clear instructions on how to reproduce, but I can instead provide the generated wpilog file, and it's converted to CSV form. 

Here is the wpilog file: [FRC_20220205_235023.zip](https://github.com/wpilibsuite/allwpilib/files/8010716/FRC_20220205_235023.zip)

Here is is, converted to csv: [output.zip](https://github.com/wpilibsuite/allwpilib/files/8010720/output.zip)
I used a custom-built tool to parse the wpilog file as well as export to CSV, but using `datalogtool` I was able to verify the total record count and the entry count as correct. If desired, I can post this as well. It's in rust, as I prefer to use a library there for implementing parsers.

I will add a couple notes on this run:
DS was connected to the robot before the program began running, and was disconnected after some time. The program did not run for 4000 seconds, and I would estimate well under 100 seconds. After a period of driving, the DS was disconnected, to allow for the logging session to halt. After which, the log file was copied off of the robot for validation and analysis.

**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1
 - RoboRIO Image: 2022.4
 - RoboRIO version: 1.0
 - C++ version: what is packaged with the above wpilib release.

sciencewhiz, 2022-02-06:
Given that Datalogger does one entry per data item logged, couldn't this type of graph happen if there is more data to log when the driver station is connected?

As a sanity check, I wrote a simple program that wrote wpi::Now to a text file in robotPeriodic. I ran for ~400 seconds in which I had the DS connected and disabled then disconnected then connected and disabled then enabled. I then did a difference between entries. The smallest I saw was 18482 and the largest was 21733, which makes sense given a 20ms nominal time for robotPeriodic.

jlmcmchl, 2022-02-07:
Perhaps - I've taken a separate look at the generated data. This photo is of our PDH voltage log - written once every 3 loops, or ~ 16hz. I'm surprised at the variability in loop timing, but that could be related to use of `std::chrono::system_clock::now()` instead of FPGA time.

[image]

I'm still perplexed by the timestamps climbing up to 4000 so quickly, as that doesn't correlate with the runtime of the robot. Was your test run in simulation or on robot? 

I'll be back into the shop on tuesday to work more - plan is to have joystick logging and state logging implemented at that time as well. I'll throw in DS connection to see if I can find any further correlation, and collect all the generated logs from that evening.

sciencewhiz, 2022-02-07:
My test was on the roboRIO.

Are you sure you're parsing the log correctly? When I run your log through datalog.py that's included in the logging PR, the largest timestamp is ~124 seconds.

jlmcmchl, 2022-02-07:
Yep, it was a bug in my parser. Timestamps look much better now, and they're not noisy. Turns out I was parsing timestamps as big endian instead of little endian, which explains why there was such a large step between 3 and 4-byte timestamps.

Thanks for your help!

closed 2022-02-07 by jlmcmchl