wpilibsuite/allwpilib #4012: SIMGUI PS4 Controller Mapping incorrect

dawonn, 2022-02-08:
The axis mapping for PS4 controllers in the Simulator GUI do not match the axis when connected to the Driver Station. I don't think this was the case last year... 

The mapping in PS4Controller.h is correct.
```cpp
  struct Axis {
    static constexpr int kLeftX = 0;
    static constexpr int kLeftY = 1;
    static constexpr int kRightX = 2;
    static constexpr int kRightY = 5;
    static constexpr int kL2 = 3;
    static constexpr int kR2 = 4;
  };
```

This is what appears to happen in the SIMGUI by default
```cpp
  struct Axis {
    static constexpr int kLeftX = 0;
    static constexpr int kLeftY = 1;
    static constexpr int kRightX = 3;
    static constexpr int kRightY = 4;
    static constexpr int kL2 = 2;
    static constexpr int kR2 = 5;
  };
```

This is what appears to happen in the SIMGUI by when 'map gamepad' is selected (whatever that is...)
```cpp
  struct Axis {
    static constexpr int kLeftX = 0;
    static constexpr int kLeftY = 1;
    static constexpr int kRightX = 4;
    static constexpr int kRightY = 5;
    static constexpr int kL2 = 3;
    static constexpr int kR2 = 4;
  };
```

sciencewhiz, 2022-02-08:
Did you select the map game pad option? 

https://docs.wpilib.org/en/stable/docs/software/wpilib-tools/robot-simulation/simulation-gui.html#adding-a-system-joystick-to-joysticks

sciencewhiz, 2022-02-08:
Map game pad is described here: https://docs.wpilib.org/en/stable/docs/software/wpilib-tools/robot-simulation/simulation-gui.html#adding-a-system-joystick-to-joysticks

PeterJohnson, 2022-02-08:
What platform is this on?  Note that on Windows you can use the real DS in simulation (check the "sim driverstation" checkbox when starting simulation.

dawonn, 2022-02-08:
I'm running simulator on linux, Driver station on windows. 

The 'map gamepad' does not have the expected mapping tho.

Daltz333, 2022-02-23:
I'm going to go ahead and close this issue. Feel free to comment if you have additional questions.

closed 2022-02-23 by Daltz333