wpilibsuite/allwpilib #4019: We can't see analog value on driver station

Ardasak, 2022-02-10:
**Describe the bug**
Hi everyone, we are trying to send data from an Arduino Leonardo to the driver station. It works all fine with sending digital values from pins but when it comes to analog values, we can't see any data on driver station. We are using a library where you can find at https://github.com/MHeironimus/ArduinoJoystickLibrary and you can see our arduino code in the screenshots part below. Also I want to mention that, we can see the analog data coming on an online tester or windows game controller tester, but we just can't see anything on driver station as its shown in the screenshots in an unbelievable way.

**Expected behavior**
We desire to see axis data on driver station and be able to access it in our code.

**Screenshots**
[Arduino Code Part 1](https://ibb.co/zSdStMB)
[Arduino Code Part 2](https://ibb.co/ZLcCGzb)
[Driver Station Image](https://ibb.co/MsK30B3)

**Desktop (please complete the following information):**
- Driver Station Version: [2022 f1]
- Arduino IDE Version: [1.8.19]

**Additional Context**
We'll be really thankful if you could help us.

ThadHouse, 2022-02-10:
The DS can only accept 6 axes, and looking at the joystick library, it seems like it defaults to 11 axes.

https://github.com/MHeironimus/ArduinoJoystickLibrary/blob/master/src/Joystick.h#L116

Try constructing the joystick object, but passing false to everything past includeYAxis.

`Joystick_ Joystick{JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK, JOYSTICK_DEFAULT_BUTTON_COUNT, JOYSTICK_DEFAULT_HATSWITCH_COUNT, true, true, false, false, false, false, false, false, false, false, false};`

Ardasak, 2022-02-11:
This worked, thank you so much man!!