wpilibsuite/allwpilib #4022: [BUG] C++ DifferentialDrive::ArcadeDriveIK Unexpected Outputs

RCoder01, 2022-02-12:
**Describe the bug**
The DifferentialDrive.arcadeDriveIK method returns a DifferentialDrive.WheelSpeeds containing unexpected values when called with an xSpeed parameter of -0.0.

**To Reproduce**
Steps to reproduce the behavior:
Java:
1. Create a Java wpilib project
2. Call DifferentialDrive.ArcadeDriveIK(-0.0, 1, false) and store/print the returned DifferentialDrive.WheelSpeeds object
3. This WheelSpeeds object will have a left attribute of -1 and a right attribute of -1

C++:
1. Create a C++ wpilib project
2. Call frc::DifferentialDrive::ArcadeDriveIK(-0.0, 1, false) and store/print the returned frc::DifferentialDrive::WheelSpeeds object
3. This WheelSpeeds object will have a left attribute of -1 and a right attribute of -1

**Expected behavior**
The method is expected to return an object with attributes left=1 and right=-1 for these inputs.

Calling the method with parameters (0.0, 1, false) returns left=1 and right=-1. Additionally, calling the method with some small but nonzero negative xSpeed parameter, like (-0.0001, 1, false) returns left=0.9999, right=-1. Notice that left is nearly 1, not the -1 observed when calling with xSpeed=-0.0.

**Screenshots**
Java:
[image]

C++:
[image]

**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1
 - OS: Windows 10
 - Java version 11
 - C++ version 17

**Additional context**
I originally ran into this problem when using RobotPy, but have now confirmed the issue is present in the officially supported languages as well.

This issue is not unique for zRotation=1, it persists when calling for any zRotation. Instead of returning one negative speed and one positive speed (like it would with xSpeed 0.0), calling with xSpeed -0.0 returns two negative values. zRotation 1 is used everywhere in this bug report for simplicity.

calcmogul, 2022-02-12:
> This WheelSpeeds object will have a left attribute of -1 and a right attribute of -1

Your own screenshots don't corroborate that, and I can't reproduce it either. The following tests pass:
```cpp
  // Left turn
  drive.ArcadeDrive(0.0, -1.0, false);
  EXPECT_DOUBLE_EQ(-1.0, left.Get());
  EXPECT_DOUBLE_EQ(1.0, right.Get());

  // Right turn
  drive.ArcadeDrive(0.0, 1.0, false);
  EXPECT_DOUBLE_EQ(1.0, left.Get());
  EXPECT_DOUBLE_EQ(-1.0, right.Get());
```
Did you negate a motor controller?

chauser, 2022-02-12:
The tests don't test for the bug -- the method is ArcadeDriveIK and the first parameter is -0.0 -- which at least in Java has negative sign. The problem in Java is line 281 of DifferentialDrive.java where the sign is copied from xSpeed but then the test on line 283 treats xSpeed as >=0 (even though it is -0.0).
Subtle bug.

calcmogul, 2022-02-12:
Looks like -0.0 == 0.0, so it assumes the first quadrant when it's actually in the second.

chauser, 2022-02-12:
Yes (-0.0==0.0) but (copySign(1.0, -0.0) == -1.0) and (copySign(1.0,0.0) ==
1.0)

PR #4028 (2022-02-18): [wpilib] Check for signedness in ArcadeDriveIK()
If xSpeed == -0.0 and zRotation > 0, the algorithm assumes it's in the
third quadrant instead of the first since +0.0 == -0.0.

Also added tests for inverse kinematic functions, fixed some
MecanumDrive test bugs, and added Java MecanumDrive.driveCartesianIK()
and KilloughDrive.driveCartesianIK() overloads with defaulted gyro angle
that C++ already had.

Fixes #4022.
commit 49adac95 (2022-02-18): [wpilib] Check for signedness in ArcadeDriveIK() (#4028)

If xSpeed == -0.0 and zRotation > 0, the algorithm assumes it's in the
third quadrant instead of the first since +0.0 == -0.0.

Also added tests for inverse kinematic functions, fixed some
MecanumDrive test bugs, and added Java MecanumDrive.driveCartesianIK()
and KilloughDrive.driveCartesianIK() overloads with defaulted gyro angle
that C++ already had.

Fixes #4022.
closed 2022-02-18 by PeterJohnson