wpilibsuite/allwpilib #4030: Build issues for allwpilib in Ubuntu 21.10

WilliamBehrens, 2022-02-13:
**Describe the bug**
I'm trying to create a minimal build in Ubuntu 21.10 but I keep getting build issues in `libuv`
[image]

**To Reproduce**
Steps to reproduce the behavior:
```
cmake -G Ninja -S . -B builddir/ \
-DBUILD_SHARED_LIBS=OFF -DBUILD_GMOCK=OFF -DINSTALL_GTEST=OFF \
-DWITH_CSCORE=ON -DWITH_EXAMPLES=OFF -DWITH_GUI=OFF -DWITH_JAVA=OFF \
-DWITH_OLD_COMMANDS=OFF -DWITH_SIMULATION_MODULES=OFF \
-DWITH_TESTS=OFF -DWITH_WPILIB=ON -DWITH_WPIMATH=ON \
-DCMAKE_BUILD_TYPE=Release
```
`cmake --build builddir --parallel`

**Desktop (please complete the following information):**
 - WPILib Version: latest main
 - OS: Ubuntu 21.10
 - C++ version gcc 11.2.0

PeterJohnson, 2022-02-13:
This is due to using a much newer compiler that warns on more things.  We have -Werror turned on so any warnings will result in build errors; we will look into fixing the warnings you're seeing, but you can work around this in the meantime by removing -Werror from the list of compile options in cmake/modules/CompileWarnings.cmake.

WilliamBehrens, 2022-02-13:
I could get it building by changing that section of CMake but I'm wanting to integrate it with the CMake fetch content API and idk if there's a way to apply a patch to remove it or some variable I can change from my CMake files

PeterJohnson, 2022-02-13:
Not that I know of.  Opened a PR to fix this specific issue as well as one to make it easier to customize target warnings (#4032).

WilliamBehrens, 2022-02-13:
Ok, thank you

PR #4031 (2022-02-14): [wpiutil] libuv: Fix sign compare warnings in gcc 11.2
Fixes #4030.
closed 2022-02-13 by WilliamBehrens