wpilibsuite/allwpilib #4040: PneumaticHub CPP API Documenation

dawonn, 2022-02-20:
https://first.wpi.edu/wpilib/allwpilib/docs/release/cpp/classfrc_1_1_pneumatic_hub.html

thhous-msft, 2022-02-20:
@dawonn Is there a specific ask here? Something you think should be updated?

ThadHouse, 2022-02-20:
@dawonn Is there a specific ask here? Something you think should be updated?

dawonn, 2022-02-20:
(Updated).

ThadHouse, 2022-02-20:
@NoahAndrews can you answer the questions above. We'll add them to the API docs, but we can pull the answers from here.

The only one I fully know the answer to is it always uses channel 0.

For min and max, I _think_ it will charge up to the max, and then discharge to the min before starting the compressor.

dawonn, 2022-02-20:
Trail goes cold here, the parameter values are passed to the PH to do... something...: 
https://github.com/wpilibsuite/allwpilib/blob/c137569f91ba387034a2ee3c82d52869bbb81b5a/hal/src/main/native/athena/REVPH.cpp#L271

I'm _guessing_ the digital or analog sensor are OR'd together to run in hybrid mode...

NoahAndrews, 2022-02-20:
Gladly! The hybrid mode turns off the compressor in either of these conditions:

* The digital pressure switch is disconnected or indicates that the tank is full
* The pressure detected by the analog sensor is greater than the specified max pressure.

The hybrid mode turns on the compressor only when the digital switch indicates that the tank is not full, and the analog sensor is below the minimum pressure.

You're correct about what the min and max pressures mean, and that only analog channel 0 can be used for compressor control.

PR #4881 (2023-01-02): Add/update documentation to PneumaticBase and subclasses (NFC)
Also update Compressor docs for consistency.

Closes #4040
closed 2023-01-02 by PeterJohnson