wpilibsuite/allwpilib #4041: Flawed command execution order

nimrod46, 2022-02-20:
Hello ppl, throughout my years here we kept using somewhat the same simple logic to stop a motor using the command-based approach:
```
execute: 
    setSpeed(1)

isFinished:
    return isBallIdentified()
```

**But this approach always introduced the same basic problem:**
The way command schedule works is[ **first to execute**](https://github.com/wpilibsuite/allwpilib/blob/4bd1f526abd499399384740bf20ed98ebcf03600/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command/CommandScheduler.java#L280) the command and **only then** checking `isFinished` which causes the motor to **stutter** in the best case, I am not sure why this is implemented like this, maybe I always missed something there.

I think it's very reasonable to assume that a command which is scheduled to run while its `isFinished` condition is already met will **only be initialized** without executing at all.

Of course, the problem is a bit more complicated than that, in recent years you introduce some major improvements which made this "bug" even **more noticeable** such as `whileActiveContinuous`, `parallelDeadlineGroup` (which also execute and then checks `isFinished`), and **new features** to come such as #4009.

Hoping to hear your thoughts about it!

Oblarg, 2022-03-15:
I'm not sure I understand the complaint here; if you don't want the command to run at all in certain cases then you can add a condition to the trigger.

nimrod46, 2022-03-19:
> I'm not sure I understand the complaint here; if you don't want the command to run at all in certain cases then you can add a condition to the trigger.

@Oblarg 
My point is that you never want to run the command if the `isFinished` condition is reached not even once, maybe it made sense to make sure `execute` is called once before `isFinished` is getting checked but today when you have built-in solutions for writing repeating commands (i.e using `whileActiveContinuous`)  I think it's only causing problems, sure you can add checks to all relevant commands but no one wants to code like this.

If the desired behavior is really to call `execute` once regardless of its `isFinished` state I think adding a call to `execute` in the initialize makes much more sense than hiding this behavior behind and affecting every type of command.

Starlight220, 2022-08-29:
I think that the `unless()` decorator (added in #4244) sufficiently provides an answer for the use case OP mentioned, and we're not looking to confusingly break a core part of how command-based has worked since its conception.
So I'm closing this; if there's anything I'm missing, reopen this or open a new issue.

closed 2022-08-29 by Starlight220