wpilibsuite/allwpilib #4045: Arcade drive os inverted after long power off, resets to normal after subsequent power off

nolanhergert, 2022-02-22:
**Describe the bug**
If we leave the robot off overnight, sometimes WPILib's Arcade Drive implementation switches the results of our inputs. (forwards and backwards <left joystick on XBox controller> becomes turning, and turning <right stick on xbox controller> becomes forwards and backwards). After a DC cycle with the same code, then things are back to normal and stay fine for subsequent DC cycles. This has occurred several times. 

**To Reproduce**
Steps to reproduce the behavior: 
While really weird, I only have the above. You can take a look at our code here: https://github.com/LibertyRobotics/FRC_2022/blob/2019-drive/src/main/java/frc/robot/RobotContainer.java

**Expected behavior**
Be able to drive the robot without flashing new code, even after sitting overnight.

**Screenshots**

**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1 (occurred on previous WPILib too)
 - OS: Windows 10
 - Java version [e.g. 1.10.2]
 - C++ version [e.g. 17]

**Additional context**

I've confirmed that it's not our xbox controller or code by printing the real-time outputs we're sending into Arcade Drive and they look correct. (Left side and right side joysticks correspond to forward/backward and turning, respectively). I can invert the variables I send to the WPILib function and things work correctly until a DC cycle. (persists through deploying new code)

I'll keep debugging it when I have a repro tomorrow. Any ideas?

jasondaming, 2022-02-22:
Does the same thing happen with other Driver Station computers?

PeterJohnson, 2022-02-22:
Check the wiring of your SparkMaxes to the motors.  Make sure there's no encoder wire breaks or phases not connected.

nolanhergert, 2022-02-22:
@jasondaming Good idea, I'll check that. 

@PeterJohnson I'll check that. I would think that it would sporadically continue to invert if it was loose, but I could be wrong too.

nolanhergert, 2022-02-22:
Just got word that this happened way back in 2019 too. So is likely unrelated with WPILib. I'll close it for now but will post a root cause if I have it.

closed 2022-02-22 by nolanhergert