wpilibsuite/allwpilib #4046: AnalogEncoder does not include an offset

LennyAbbas, 2022-02-22:
If the encoder is not mounted perfectly then there will be an offset between the 'actual' zero angle and the angle that the analog input thinks is zero. AnalogEncoder already has a reset method which 'zeroes' the analog input, but if the offset is constant (ie doesn't change periodically) then I don't want to have to call reset() every time I turn on the robot. For example if one is using analog encoders as the turning encoders for the modules in a swerve robot, and the encoders are mounted with an offset, one would need to manually put all the wheels straight and then call reset() every time the robot is turned on.

I think the AnalogEncoder constructor should have an option to take in an offset (in volts, which would default to 0 if not set). It would then use the offset to return an accurate encoder reading when get() is called.

PeterJohnson, 2022-02-22:
You're essentially just looking for a setPositionOffset()?

LennyAbbas, 2022-02-22:
also adding another method to set whether it is inverted might be useful

PeterJohnson, 2022-02-22:
The equivalent of Encoder.setReverseDirection()?

ThadHouse, 2022-02-22:
Reversing an analog encoder isn't really possible. Because the analog voltage is a fixed direction, and the counting has to be in sync with the voltage.

Also, this isn't something that would be blocking. Theres nothing stopping you from adding the offset yourself, which is exactly what we would be doing.

ThadHouse, 2022-02-22:
Also, this is a duplicate of https://github.com/wpilibsuite/allwpilib/issues/3188

LennyAbbas, 2022-02-22:
#3188 is labelled as "state: approved" so does that mean we can close this issue because this has already been decided on?

PR #4052 (2022-02-25): Some API and behavior fixes to absolute encoders
SetPositionOffset was added. Been requested multiple times, and easy to implement.

The javadocs mentioned GetPositionInRotation. It has tripped up many people how to get the absolute position from the encoder (You currently have to have precreated the DutyCycle object). Add this method (as GetAbsolutePostition) to make this easier to do.

The checks for making sure a matching set of values was read was doing direct double comparisions. This worked ok in the DutyCycle case, but has problems in the analog case. Solve this by using an epsilon comp [... cut]
commit bc9e96e8 (2022-02-25): [wpilib] Absolute Encoder API and behavior fixes (#4052)

SetPositionOffset was added. Been requested multiple times, and easy to implement.

The javadocs mentioned GetPositionInRotation. It has tripped up many people how to get the absolute position from the encoder (You currently have to have precreated the DutyCycle object). Add this method (as GetAbsolutePostition) to make this easier to do.

The checks for making sure a matching set of values was read was doing direct double comparisions. This worked ok in the DutyCycle case, but has problems in the analog case. Solve this by using an eps [... cut]
closed 2022-02-25 by PeterJohnson