wpilibsuite/allwpilib #4047: WPI motor controller's stopMotor() api break follow() configurations for slave motors.  Which means "Test" mode reset any following motors configuration to follow.

wegnergit, 2022-02-22:
**Describe the bug**
WPI motor controllers (i.e. WPI_TalonFX, WPI_VictorSPX, WPI_TalonSRX, ...) clear follow() configuration.
Their stopMotor() api breaks follow() configurations for slave motors.  
Which means "Test" mode will reset any following motors configuration to follow, so if reenter "Teleoperated" or "Autonomous" mode and after "Test" mode none of follow motors will move.

**To Reproduce**
Steps to reproduce the behavior using "Test" mode:
1. Run simulator (with joystick 0 setup and expand devices TalonFX [1] and [2]) or robot 
2. Start in Teleoperated mode
3. See motor 1 (mater) and 2 (slave) running percentOutput at 1.0
4. Switch to Test mode -> See only master motor at speed -- percentOutput on 1 at 1 and on 2 at 0.0!!!!! (ISSUE slave motor no longer following)    

Steps to reproduce the behavior using example robot code (using commands to show it is WPI's stopMotors() method causing issue):
1. Run simulator (with joystick 0 setup and expand devices TalonFX [1] and [2]) or robot 
2. Start in Teleoperated mode
3. See motor 1 (mater) and 2 (slave) running percentOutput at 1.0
4. Press A button -> See motors stop -- both motors percentOutput at 0.0
5. Press B button -> See only master motor at half speed -- percentOutput on 1 at about .478 and on 2 at 0.0!!!!! (ISSUE slave motor no longer following)
6. Press X button -> see slave motor back to following (only after reconfiguring slave motor to follow -- refollowMotors()) 

**Expected behavior**
The slave motors configuration should be cleared after stopMotor() is executed and should not have to be reconfigured to follow (i.e. slaveMotor.follow(masterMotor);)

**Screenshots for second scenario**
After Teleoperated mode:
[image]
After Press A button:
[image]
After Press B button (causes issue stopMotors() called):
[image]
After Press X button ( reconfigures follow motor fixes following but should not have to do this):
[image]

 
**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1
 - OS: Windows 10
 - Java version 1.11 (not sure)
 - C++ version N/A (may exist with C++ too)

**Additional context**
Attaching zip file with example FRC command base robot program (WHYNOTFOLLOW) to reproduce issue as described above (see RobotContainer.java and ExampleSubsystem.java)
[src.zip](https://github.com/wpilibsuite/allwpilib/files/8119274/src.zip)
.

PeterJohnson, 2022-02-22:
This should be reported to CTRE, those classes (WPI_TalonSRX etc) are part of their vendor library, not WPILib.

closed 2022-02-22 by PeterJohnson

wegnergit, 2022-02-22:
Thank you.  Sorry for submitting to wpilib and not CTRE.

wegnergit, 2022-02-22:
FYI, I created issue with CTRE https://github.com/CrossTheRoadElec/Phoenix-Releases/issues/28.  They believe this is working as intended and you will to reconfigure to follow each time stopMotor() is called.