wpilibsuite/allwpilib #4048: `HolonomicDriveController` should set the theta PID controller to continuous input

jonahsnider, 2022-02-22:
**Is your feature request related to a problem? Please describe.**

`HolonomicDriveController` does not set the profiled PID controller for controlling your robot's heading (theta) to use continuous input (ie. 360° is the same as 0°).

Additionally, [the docs page for it](https://docs.wpilib.org/en/stable/docs/software/advanced-controls/trajectories/holonomic.html) doesn't mention that you'll need to set the controller as continuous.

**Describe the solution you'd like**

`HolonomicDriveController` should automatically set the provided theta PID controller as using continuous input, or at the very least, have a blurb on the docs page to remind users to do so themselves.

Starlight220, 2022-10-02:
Do we want to make this change before beta?

calcmogul, 2022-10-02:
Seems like an important fix, so I think it should be included in beta.

PR #4453 (2022-10-06): [wpimath] Enable continuous angle input for HolonomicDriveController
Fixes #4048
closed 2022-10-06 by PeterJohnson