wpilibsuite/allwpilib #4062: Scheduler execution order causes undesired behavior

rzblue, 2022-02-27:
**Describe the bug**
This issue has to do with the order `CommandScheduler` runs its functions and how that interacts with `ProfiledPIDSubsystem` and likely `PIDSubsystem`

The Scheduler runs its functions in the following order:
1.    Subsystem periodic methods
2.    Run buttons (schedule new commands, runs `initialize()`)
3.    Run `execute()` on commands
4.    Check `isFinished()`, remove finished commands

If a command calls `setGoal()` on a ProfiledPIDSubsystem during `initialize()` or `execute()`, the goal will not be updated with the internal controller until the next call of `periodic()`. 
This means if you use `getController.atGoal()` for the `isFinished()` method, the internal controller will not have the updated goal, and will return based on the previously set goal, likely ending the command.

**To Reproduce**
Implement ProfiledPIDSubsystem.
Implement a command that calls `subsystem.setGoal()` during `initialize` and uses `subsystem.getController().atGoal()` for `isFinished()`
Schedule the command with a new goal.

The command will likely finish immediately due to the fact that `atGoal` will return a value based on the previous goal, since the controllers goal is not updated with the subsystem's goal until the next call to `periodic()`.

**Expected behavior**
Command runs until the subsystem is actually at its desired goal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: 2022.4.1
 - OS: Windows 11
 - Java version: openjdk 17.0.1 2021-10-19
 - C++ version n/a

**Additional context**
Add any other context about the problem here.

PeterJohnson, 2022-02-28:
Duplicate of #4041?

rzblue, 2022-02-28:
Don't think so, that one is referencing how on a command that is already finished (isFinished returns true when the command is started), execute is still called

Starlight220, 2022-03-15:
Shouldn't `setGoal` call `setSetpoint` on the backing controller object? I don't remember if that's in the current impl, but it should be.
Changing `CommandScheduler`'s run loop order sounds like a very drastic, breaking, and unproportionate solution for this issue.

rzblue, 2022-03-15:
Currently calling `setGoal` sets an internal goal variable, which is then used during `periodic` when the backing controller's `calculate` method is called.

I think the internal goal variable should be removed and the backing controller should be allowed to track the goal by delegating the `setGoal` method to it.

I agree, `CommandScheduler` shouldn't be rewritten here. Sorry if it sounded like that's what I wanted.

PR #4414 (2022-09-17): Fix ProfiledPIDSubsystem setGoal behavior
Fixes #4062
closed 2022-09-17 by PeterJohnson