wpilibsuite/allwpilib #4083: SlewRateLimiter has no method to discover the rate limit for the instance

robagale, 2022-03-09:
I can initialize a SlewRateLimiter with a specific rate to control the rate of change of an input value. However I can't interrogate the instance later to discover the rate limit in use. If I want to know the rate later I have to maintain a variable to handle it.  

Repro steps:
1. Create a SlewRateLimiter instance in a class such as your drivetrain subsystem:
`m_driveRateLimiter = new SlewRateLimiter(DriveConstants.kDriveRateLimit);`
2. Set up a mechanism such as a Dashboard field to supply a new rate value at runtime
3. Attempt to check the current value against the supplied value to see if a new instance needs to be created
```
    if (driveSlew != m_driveRateLimiter.getRate()) {
      m_driveRateLimiter = new SlewRateLimiter(driveSlew, m_lastDriveInput);
    }
```
This code is not valid as getRate() does not exist.

**Expected behavior**
The above getRate() method is available

To get around this limitation I subclassed SlewRateLimiter and implemented getRate() in the subclass. However SlewRateLimiter already maintains a private field m_rateLimit and making it available to read would be a trivial change.

**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1
 - OS: Windows 10

calcmogul, 2022-03-09:
We generally avoid getters for things the user already has access to. In this case, the slew rate was a constructor argument.

Starlight220, 2022-11-23:
What's our plan forward with this?

closed 2023-08-03 by calcmogul