wpilibsuite/allwpilib #4088: Kalman Filter Incorrectly Requires 1:1 Predict:Update

Shaun345, 2022-03-12:
**Describe the bug**
The Kalman Filter implementation requires 1:1 relationship between predict and update.

**To Reproduce**
By inspection

**Expected behavior**
Kalman filter should allow for multiple predictions between corrections. Each prediction should update the covariance matrix per the algorithm.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - WPILib Version: 2022.4.1
 - Java version 11.0.13

**Additional context**
Found on java, likely needs C++ fix.

calcmogul, 2022-03-12:
The most common use case for KalmanFilter is symmetric updates, so this was a performance optimization. I suppose we could generalize it a bit more.

jee7s, 2022-03-13:
Disagree. There are many KF applications where a fast sensor is used to estimate state through prediction while a slow sensor with lower covariance "trues up" the state via a KF update. Real world example is IMU + GPS. FRC example would be IMU + Odometry or IMU/Odometry + distance sensors at known locations.

I would also suggest the KalmanFilter should be the most generalized implementation that is canonical with the filter definition as presented in the literature. A SymmetricKalmanFilter (or some such class) could contain the special case where the ratio of predict to update is 1:1.

calcmogul, 2022-03-13:
But all the examples you listed are for nonlinear systems. The EKF and UKF we have support time-varying Kalman gains, of course. The only linear systems in FRC are the flywheel, elevator, and arm (ignoring gravity via a feedforward), and those don't need asymmetric updates.

jee7s, 2022-03-13:
In the steady state kalman gain case, that is in effect a weighted average filter and is rightly justified for use when estimating things like a flywheel. As I mention, I think there's a perfectly valid case for a SymmetricKalmanFilter that implements this as opposed to a KalmanFilter which should follow the canonical implementation.

Not to get into a contest here, but linear position estimation using a distance sensor and accelerometer would be a valid FRC use case that would require multiple predictions per update, depending on the relative sampling frequency of each sensor.

I would recommend caution around premature optimization for special cases that break the consumer's expectation of behavior. I think this is especially true in cases where something broadly understood (like the KF) is involved. The documentation related to the KF makes no mention of this optimization and related constraint.

calcmogul, 2022-03-13:
> I would recommend caution around premature optimization for special cases that break the consumer's expectation of behavior. ... The documentation related to the KF makes no mention of this optimization and related constraint.

Fair. I wouldn't consider it premature optimization though. For the systems we expected users to use at the time of writing that class, there wasn't a need for asymmetric updates, and floating point can and does lead to stability issues.

For example, in our UKF's sigma point generation, we have to take the Cholesky decomposition of the error covariance P. The naive UKF implementation commonly has P become either asymmetric or no longer positive definite over time, which makes the filter diverge (we've hit this problem several times in the wild). The typical fix for this is to run the square root form of the UKF, but we haven't gotten around to implementing it because the paper for it is difficult to read.

For a KF, filter divergence usually occurs when the error covariance entries grow several orders of magnitude apart. If we make the KF use a time-varying Kalman gain, I'd prefer we use Pₖ₊₁⁺ = (I−Kₖ₊₁C)Pₖ₊₁⁻(I−Kₖ₊₁C)ᵀ + Kₖ₊₁RₖKₖ₊₁ᵀ instead of Pₖ₊₁⁺ = (I−Kₖ₊₁C)Pₖ₊₁⁻ since it's more numerically stable. We should do that in the EKF too.

PR #5951 (2023-11-23): [wpimath] Make KalmanFilter variant for asymmetric updates
Fixes #4088.
closed 2023-11-23 by PeterJohnson