wpilibsuite/allwpilib #4090: Add simple intersection computations for simulation purposes

virtuald, 2022-03-12:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to easily answer the following question: Is a particular point on my robot overlapping with a rectangle on the field (for example, the climbing bar).

**Describe the solution you'd like**
A set of simple shapes/points that interact with the existing wpimath geometry classes to allow determining if things intersect with each other. Obviously there's a lot of different kinds of intersections, so probably only a few useful ones would need to be implemented:

* point within rectangle 
* point within circle/oval
* rectangle/robot overlaps rectangle 

Might be useful for Mechanism2d to be able to display these planes (maybe the intersection logic should live there?)

OpenCV has a bunch of this logic I think. Probably other math libraries too.

**Describe alternatives you've considered**
* Write it myself and mess it up.
* Use an external library

calcmogul, 2022-03-12:
We can reuse the logic from the EllipticalRegionConstraint and RectangleRegionConstraint for the first two points. The third could implement this paper: https://www.cs.jhu.edu/~misha/Spring20/ORourke82.pdf

virtuald, 2022-03-12:
A good POC would be integration with arm simulation / elevator simulation examples to have a DIO turn on/off when the arm is at a particular angle, or elevator thing is at a particular height.

calcmogul, 2022-03-12:
For arm and elevator simulations, these shape intersections are overkill. Teams can easily just do
```
if (abs(sim state - sensor position) < 1e-9) {
  turn the DIO on
} else {
  turn the DIO off
}
```

auscompgeek, 2022-12-18:
A good example of 2D collision detection I'd like to see: determining when my robot has picked up a gamepiece during an autonomous routine.

PR #6555 (2024-06-05): [wpimath] Add geometry classes for Rectangle2d and Ellipse2d
Resolves #6446, resolves #4090 

Adds Rectangle2d and Ellipse2d classes, for checking if points intersect certain zones. The Java implementation includes tests, structs, protobufs, and javadocs. 

This also adds a `rotateAround(Translation2d other, Rotation2d rot)` method for rotating a Translation2d around another point (not the origin) by a certain amount. This is used in checking for intersection/containment of translation2ds in a rotated 2d area (ex, if checking if a point is contained in a rotated rectangle, the point is rotated [... cut]
closed 2024-06-05 by PeterJohnson