wpilibsuite/allwpilib #4102: Mechanism2dRoot should have convenience methods for manipulating children

virtuald, 2022-03-15:
If one creates a large number of ligaments that aren't really doing anything, it would be convenient if I only had to keep the root around.

In particular, SetColor/SetLineWeight would be nice.

PeterJohnson, 2022-03-16:
Mechanism2d is designed for visualization of robot mechanisms: eg single/2-jointed arms, elevators, etc.  It’s not designed to be a generic drawing canvas (although clearly it can be abused to do that).  Instead of trying to turn it into a generic canvas, maybe propose a new Canvas Sendable element with an appropriate API?

virtuald, 2022-03-20:
I think it would be better (but similar) to define a Canvas sendable, and build the mechanism2d on top of it. The key is it would be good to be able to draw what you're simulating.

... I was going to post a gif of something I'm working on to illustrate what I'm trying to accomplish, but apparently recording OpenGL on Linux is annoying and doesn't seem to work.

FWIW, these are the helpers I'm using at the moment:

```python
@dataclasses.dataclass
class Shape:
    root: wpilib.MechanismRoot2d
    items: typing.List[wpilib.MechanismLigament2d]

    def setPosition(self, x: float, y: float):
        self.root.setPosition(x, y)

    def setColor(self, c: wpilib.Color8Bit):
        for item in self.items:
            item.setColor(c)

class Mechanism(wpilib.Mechanism2d):
    def __init__(self, width: float, height: float) -> None:
        super().__init__(width, height)

    def make_point(self, name: str, cx: float, cy: float, sz: int = 10) -> Shape:
        root = self.getRoot(name, cx, cy)
        return self._make(root, 0.50, 0, 1, sz)

    def make_triangle(
        self, name: str, cx: float, cy: float, side: float, line_width=6
    ) -> Shape:
        x = cx + side / 2
        y = cy - side / 2
        root = self.getRoot(name, x, y)
        return self._make(root, side, 120, 3, line_width)

    def make_hex(
        self, name: str, x: float, y: float, side: float, line_width=6
    ) -> Shape:
        # x, y are near bottom right corner
        root = self.getRoot(name, x, y)
        return self._make(root, side, 60, 6, line_width)

    def _make(
        self,
        root: wpilib.MechanismRoot2d,
        side: float,
        angle: float,
        n: int,
        line_width,
    ) -> Shape:
        item = root
        items = []
        for i in range(n):
            item = item.appendLigament(f"{i}", side, angle, line_width)
            items.append(item)

        return Shape(root, items)
```

virtuald, 2022-03-20:
[image]

virtuald, 2022-04-11:
Finally got around to making a video of it: https://youtu.be/MHMJc4h1hnw