wpilibsuite/allwpilib #4109: Add position delta odometry for swerve and mecanum

calcmogul, 2022-03-17:
Position delta odometry is more accurate than velocity odometry. Differential drive has position delta odometry, but swerve and mecanum have velocity odometry.

We didn't use position deltas originally because it didn't fit the API shape given we needed velocity kinematics for other things. Differential drive was simple enough that the position delta kinematics were written inline. The equations for position delta and velocity kinematics look the same, but the names didn't fit both. The resolution may be adding duplicates of the classes but with different names to reflect the data that goes in them.

The only modification needed in the odometry class is not using a time delta to turn the velocities into position deltas.

connorworley, 2022-04-27:
Planning on doing this unless someone else is already on it.

calcmogul, 2022-04-27:
Go ahead.

closed 2022-10-26 by calcmogul