wpilibsuite/allwpilib #4110: Constructors to command groups don't play nice with other command groups, fail with unhelpful error message

Mee42, 2022-03-17:
**Describe the bug**
Run the two code examples, one should compile and one should not compile

**To Reproduce**
```cpp
// compiles
new frc2::ParallelDeadlineGroup(frc2::PrintCommand("."), *(new frc2::InstantCommand([ ]{   })));
// does not compile
new frc2::ParallelDeadlineGroup(frc2::PrintCommand("."), *(new frc2::SequentialCommandGroup()));
```
<details>
  <summary>Compiler Error</summary>
<pre><code>
 > Task :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
In file included from /home/carson/wpilib/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0/bits/locale_conv.h:41:0,
                 from /home/carson/wpilib/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0/locale:43,
                 from /home/carson/.gradle/caches/transforms-3/b4b8fedce584c8e39150e45c17839c01/transformed/wpimath-cpp-2022.4.1-headers/units/base.h:82,
                 from /home/carson/.gradle/caches/transforms-3/b4b8fedce584c8e39150e45c17839c01/transformed/wpimath-cpp-2022.4.1-headers/units/angle.h:29,
                 from /home/carson/.gradle/caches/transforms-3/b4b8fedce584c8e39150e45c17839c01/transformed/wpimath-cpp-2022.4.1-headers/units/math.h:31,
                 from /home/carson/.gradle/caches/transforms-3/960cdbb10f866f71576bf87d87cb04de/transformed/wpilibc-cpp-2022.4.1-headers/frc/TimedRobot.h:12,
                 from /home/carson/robotics/RapidReact/src/main/include/Robot.h:7,
                 from /home/carson/robotics/RapidReact/src/main/cpp/Robot.cpp:5:
/home/carson/wpilib/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0/bits/unique_ptr.h: In instantiation of 'typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = frc2::SequentialCommandGroup; _Args = {frc2::SequentialCommandGroup&}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<frc2::SequentialCommandGroup, std::default_delete<frc2::SequentialCommandGroup> >]':
/home/carson/.gradle/caches/transforms-3/ca7a01ec830179f35a4b5609e774838f/transformed/wpilibNewCommands-cpp-2022.4.1-headers/frc2/command/ParallelDeadlineGroup.h:79:77:   required from 'void frc2::ParallelDeadlineGroup::AddCommands(Types&& ...) [with Types = {frc2::SequentialCommandGroup&}; <template-parameter-1-2> = void]'
/home/carson/.gradle/caches/transforms-3/ca7a01ec830179f35a4b5609e774838f/transformed/wpilibNewCommands-cpp-2022.4.1-headers/frc2/command/ParallelDeadlineGroup.h:63:5:   required from 'frc2::ParallelDeadlineGroup::ParallelDeadlineGroup(T&&, Types&& ...) [with T = frc2::PrintCommand; Types = {frc2::SequentialCommandGroup&}; <template-parameter-1-3> = void; <template-parameter-1-4> = void]'
/home/carson/robotics/RapidReact/src/main/cpp/Robot.cpp:163:97:   required from here
/home/carson/wpilib/2022/roborio/arm-frc2022-linux-gnueabi/usr/include/c++/7.3.0/bits/unique_ptr.h:825:30: error: use of deleted function 'frc2::SequentialCommandGroup::SequentialCommandGroup(frc2::SequentialCommandGroup&)'
     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/carson/robotics/RapidReact/src/main/include/subsystems/Intake.h:7:0,
                 from /home/carson/robotics/RapidReact/src/main/include/Robot.h:12,
                 from /home/carson/robotics/RapidReact/src/main/cpp/Robot.cpp:5:
/home/carson/.gradle/caches/transforms-3/ca7a01ec830179f35a4b5609e774838f/transformed/wpilibNewCommands-cpp-2022.4.1-headers/frc2/command/SequentialCommandGroup.h:68:3: note: declared here
   SequentialCommandGroup(SequentialCommandGroup&) = delete;
   ^~~~~~~~~~~~~~~~~~~~~~
</code></pre>
</details>

**Expected behavior**
For some reason, it just does this. I don't know C++ well enough to debug the issue. It's something with it being an lvalue of a sequencial command group in specific.

At the very least, if this is intended behavior, there should be a __clear__ error for users of WPILib to see, not what's currently generated

**Desktop**
 - WPILib Version: 2020.4.1
 - OS: Arch Linux, Kernel 5.16.0-arch1-1
 - Java version: I've got 17.0.1+12, 1.8.0_312-b10, and 11.0.13+8.  Latter is the default (`java` and `JAVA_HOME`)
 - C++ version:  C++17 supposedly

Mee42, 2022-03-17:
SequentialCommandGroup is made not to be copied. So it doesn't copy.

closed 2022-03-17 by Mee42

virtuald, 2022-03-17:
For future code: My understanding is that you generally shouldn't ever be using the `new` keyword at all with C++ Commands.