wpilibsuite/allwpilib #4117: The problem when using *Manage Vendor Libararies*

mendax1234, 2022-03-20:
**Describe the bug**
A clear and concise description of what the bug is.

l try to add PathPlannerLib to our code. But it fails.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. ...
2. ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: [2022.4.1]
 - OS: [windows10 pro]
 - Java version [e.g. 1.10.2]
 - C++ version [e.g. 17]

**Additional context**
Add any other context about the problem here.

Starlight220, 2022-03-20:
Can you visit that URL in a browser? It's likely a proxy or other cert issue.

Just to isolate the issue, can you try adding the vendordep using the [CLI](https://docs.wpilib.org/en/latest/docs/software/vscode-overview/3rd-party-libraries.html?highlight=vendordep#command-line)?

mendax1234, 2022-03-20:
Yes, l can visit the URL in browser.

l can't do it using the CLI

Starlight220, 2022-03-21:
If you can see the JSON content in the browser, you can just copy it into a file in the `vendordeps/` directory.

mendax1234, 2022-03-21:
After doing this , l can't build my code. 

[image]

mendax1234, 2022-03-21:
Now , it's okay. It's a problem of proxy.

closed 2022-03-21 by mendax1234