wpilibsuite/allwpilib #4130: Reset(Rotation2d angle) overload for gyroscope and IMU classes

calcmogul, 2022-03-26:
**Is your feature request related to a problem? Please describe.**
When using a gyroscope for drivetrain control, a separate heading offset needs to be maintained when the odometry is reset for use with the drivetrain controller.

**Describe the solution you'd like**
Ideally, the gyroscope class would maintain an offset internally, and `Reset(Rotation2d angle)` would call `Reset()` and override the offset.

**Describe alternatives you've considered**
The odometry class could return the offset it has internally, but that doesn't help people who aren't using the odometry class. Since this is a gyro problem, it should be solved in the gyro class.

Starlight220, 2022-03-27:
Wouldn't the offset functionality in the odometry classes conflict with the one you're suggesting to add to the gyro classes? Or are you suggesting to remove it from the odometry classes?

calcmogul, 2022-03-27:
It actually doesn't conflict. 3512 added their own offset variable to all gyro angles, and the odometry class worked fine with ResetPosition().

Starlight220, 2022-03-27:
How doesn't it conflict? As long as only one is reset, sure; but if both are reset then I don't see how the offsets won't accumulate/conflict.

calcmogul, 2022-03-27:
ResetPosition() computes its gyro offset from whatever gyro angle you give it, so if Reset(Rotation2d) included an offset already, then the odometry offset will inherently take that into account. ResetPosition() is mapping a gyro angle to a field-relative pose, so the actual gyro heading itself doesn't matter in the mapping operation.

Gold856, 2023-11-12:
Since the gyro and accelerometer interfaces have been deprecated, can this issue be closed? Or should these methods still be added to the applicable classes individually?

calcmogul, 2023-11-12:
It can be added to individual classes.