wpilibsuite/allwpilib #4133: Document DataLog value and entry ordering guarantees, if any

calcmogul, 2022-03-28:
Values are appended to entries in the order they're given to the entry. Is an entry written to the log file when finish() is called? Are entries written atomically to the log file? I assume there's no guaranteed log file ordering when entries are written from different threads.

PeterJohnson, 2022-03-28:
Finish() is used only to indicate in the log that a particular entry is no longer being used / release the mapping.  Finish() is *not* required to be called for data to be written (entries in the log are written as Append() calls are being made), and in fact is not required to be called at all--to avoid shutdown races where the DataLog might be destroyed before other objects, we generally don't call Finish() from destructors, for example.

For threading, it's a typical multiple-supplier, single-consumer setup.  So calls are atomic, as are writes to the log, but there is no guaranteed order between threads (whichever thread grabs the write mutex first will get written first).

I'll plan on adding the above to the class documentation.  Any additional clarification?

calcmogul, 2022-03-28:
I think that sufficiently describes what the log file structural invariants are.

PR #4140 (2022-03-29): [wpiutil] DataLog: Document finish and thread safety
Fixes #4133.
closed 2022-03-29 by PeterJohnson