wpilibsuite/allwpilib #4141: Trajectory constructor crashes when given an empty vector of states

romi2002, 2022-03-30:
**Describe the bug**
Whenever an `frc::Trajectory` is created with an empty vector, the program crashes when calling the constructor.

**To Reproduce**
Steps to reproduce the behavior:
```
std::vector<frc::Trajectory::State> states;
frc::Trajectory traj(states);
```
**Expected behavior**
An exception should be raised instead of segfaulting when trying to get the total time.

ysthakur, 2022-05-17:
Both Java and C++ have constructors to create empty trajectories. Is there a legitimate use for them?

Starlight220, 2022-05-17:
> We want to be able to initialize Trajectorys in a delayed manner, but Sample() can't be given a sane behavior before the user does that.

(@calcmogul, https://github.com/wpilibsuite/allwpilib/pull/3184#issuecomment-797826915) 

I do think that if the constructors aren't intended for regular team use then they shouldn't necessarily be that visible: instantiating `Trajectory` objects through `TrajectoryGenerator` isn't the most intuitive. Maybe the constructors should be made private, and expose factory methods instead?

calcmogul, 2022-05-17:
Being able to pass in a vector of States is an intended use case. It lets users generate custom trajectories.

Starlight220, 2022-05-17:
Yes, but as it's intended as an escape hatch rather than the common use form, having it as a factory method makes more sense to me.

calcmogul, 2022-05-17:
Making it a factory method or a constructor makes no difference semantically.

Starlight220, 2022-05-17:
I disagree. The intuitive way to construct an object (at least in Java) is via the constructor, not via a factory method in a utility class.  Therefore, the constructor is more discoverable.

calcmogul, 2022-05-17:
To make a trajectory that's actually useful, you have to use TrajectoryGenerator, so users already know about it from https:://docs.wpilib.org.

Starlight220, 2022-05-18:
If that constructor isn't the primary way to get a `Trajectory`, it should be clearly indicated (at least in the API docs) and point to whatever *is* the primary way to get an instance.

calcmogul, 2022-05-18:
https://docs.wpilib.org/en/stable/docs/software/advanced-controls/trajectories/trajectory-generation.html#generating-the-trajectory

Starlight220, 2022-05-18:
I meant API docs. We shouldn't expect users to constantly have frc-docs open in the background.

PR #5497 (2023-08-03): [wpimath] Make Trajectory throw on empty lists of States from users
Fixes #4141.
commit 6db2c429 (2023-08-03): [wpimath] Trajectory: Throw on empty lists of States (#5497)

Fixes #4141.
closed 2023-08-03 by PeterJohnson