wpilibsuite/allwpilib #4148: Subsystems should support the same periodic and init methods as `IterativeRobotBase`

jonahsnider, 2022-04-04:
**Is your feature request related to a problem? Please describe.**

`IterativeRobotBase` has methods for various init modes (ex. `teleopInit()`) and periodic methods (ex. `autonomousPeriodic()`).

Subsystems do not have the same methods. They only have `periodic()` which is effectively the same as `IterativeRobotBase#robotPeriodic()`.

**Describe the solution you'd like**

Subsystems should support the same periodic and init methods as `IterativeRobotBase`.

**Describe alternatives you've considered**

A workaround for now is to just expose methods in subsystems, pass them up to `RobotContainer`, expose them there, and then pass them to `Robot` where they're actually called.

You can also use the `Driverstation` methods to check if the robot is/isn't in a given state in `Subsystem#periodic()`. (ex. `DriverStation.isTeleopEnabled()`)

**Additional context**

https://discord.com/channels/176186766946992128/528555967827148801/960652141159125053

Starlight220, 2022-04-04:
Expansion of #3442.

> You can also use the Driverstation methods to check if the robot is/isn't in a given state in Subsystem#periodic(). (ex. DriverStation.isTeleopEnabled())
The `RobotState` class exists for that too.

Starlight220, 2024-01-15:
Another request for `simInit` in #6228:

> **Is your feature request related to a problem? Please describe.**
> Currently there only exists a method to periodically update simulation objects. This requires the creation of simulation objects either regardless of if the robot is in simulation,  or extra logic to check if it is in simulation before creating these objects.
> 
> **Describe the solution you'd like**
> It would be nice if there was an additional overridable method added to Subsystems that only runs once at robot start when the robot is indeed being simulating, allowing for simulation specific object to be Initialized. 
> 
> If this is a reasonable feature to add I have no problem creating a PR to add this feature. 
>

Starlight220, 2024-01-15:
In general, the problem to do this is that `Subsystem` is currently used for multiple different things, where it's intended to be primarily for hardware mutexing.
Adding more methods increases the significance of other purposes, which isn't necessarily something we want to do.

Additionally, adding these methods to the subsystem in team code is quite easy, and even more so in the discussed (2025?) inlining of RobotContainer into Robot.
As for `simInit`, an `if (RobotBase.isSimulation)` block at the end of the constructor accomplishes the task without much verbosity and with more flexibility for hardware-attached simulation.

Gold856, 2026-08-24:
Mechanism in Commands v3 doesn't have a periodic method at all.

closed 2026-08-24 by Gold856