wpilibsuite/allwpilib #4149: Improve `Color#toString()`/`Color8Bit#toString()` in Java

jonahsnider, 2022-04-04:
**Is your feature request related to a problem? Please describe.**

Colors can be stringified in a human-friendly way, but they aren't. They use the default Java `Object#toString()` behavior of returning the simple class name with the memory address appended (easy to mistake for a hex color code!).

**Describe the solution you'd like**

Named colors should get special `#toString()` values:
```java
Color.kBlack.toString(); // "kBlack"
```

Other colors should get

```java
new Color8Bit(0xff, 0xbb, 0xcc); // "#ffbbcc"
```

**Additional context**

https://discord.com/channels/176186766946992128/368993897495527424/960415849259368468

auscompgeek, 2022-04-05:
Mapping named colours back to their names would probably be a lot of work. Would just returning the hex code be reasonable too?

On a similar note, in RobotPy we have a `__repr__` implementation for colours, which give something like `Color8Bit(255, 187, 204)` (i.e. it returns an expression to reproduce the object, as is standard behaviour for `repr()` in Python). I think a differing `toString` would be reasonable too.

I don't think this should be exclusive to Java - no reason C++ shouldn't get a similar method.

Starlight220, 2022-06-03:
The PR that closed this fixed the confusion, but doesn't deal with names for specialized colors.

prateekma, 2022-06-03:
It's just a variable, you can't really do that without reflection (not even sure if you can do it with that).

calcmogul, 2022-06-03:
You could compare the color components with kBlack, for example, then print "kBlack" if they match.

PeterJohnson, 2022-06-03:
If you write a big switch statement, the compiler will do most of the work for you.  But I agree that it might be more expensive than you might want to pay if you're e.g. logging the color to a CSV file every iteration or something like that.  Maybe not include it in toString(), but add a function like toPrettyString()?

Starlight220, 2022-06-05:
I was thinking of adding a `string` constructor parameter that defaults to the RGB hex. That way it's cached for all objects.

Starlight220, 2022-06-06:
That works too.

PR #4286 (2022-06-03): [wpilibj] Add toString() methods to Color and Color8Bit
Closes #4149

Let me know if there is some other desired convention that we should use for toString()
closed 2022-10-04 by PeterJohnson