wpilibsuite/allwpilib #4152: Gryo display in fieldsim has quit working

Crossle86, 2022-04-07:
Our code has a gryo and we track the direction the robot is pointing. The code posts this to the networktables via SendableBuilder creating a data block named Gryo2 with type Gyro. We then have a widget on shuffleboard that displays the value posted to networktable entry. This works and has worked for quite some time.

So along comes sim and on the fieldsim display gui, under network tables/Smartdashboard drop down I see and select Gyro2 and a nice widget appears with the 360 degree display following our simulated gyro. All good been working great...

Until sometime in the last couple of days (not sure as the gyro widget was overlayed by other widgets) that gyro widget no longer displays the 360 degree display and just has the text "unknown gyro" on it's display surface. No clue what has changed. Again, the data is present in the networktables and looks just like it has since we started doing this so it seems something has changed on the simgui end.

Java project on 2022.4.1

Crossle86, 2022-04-16:
Closing issue as it went away by itself...

Crossle86, 2022-11-23:
So this problem returned and is present under 2022 and 2023 sim guis. So the sendable is datatype Gyro and the data field is "Heading". The widget displayed on the sim gui used to display a nice 360 degree dial with pointer that followed the value returned by the function linked to the name Heading. Just quit and while it may have come back last April per my comment above, it went away again and has remained gone. The widget that appears is just a blank box with the text "Unknown Gyro".

Related to this, I have not found any doc of what the valid values are for the SendableBuilder.setSmartDashboardType function. Best I can tell is datatype "Gyro" should be ok.

PeterJohnson, 2022-11-23:
Most likely this is the same issue that #4674 fixes (not yet in a beta release).  A workaround in the meantime would be to have another NT client (OutlineViewer or another dashboard) connect to the sim--if that fixes the issue, it will confirm it's the same issue.

Crossle86, 2022-11-23:
So your mention of "Value" caused me to turn on LW and look at what it displayed for my navx. There, the data for the heading is named "Value". So from that I assumed the gyro widget was looking for the heading data on an item ("topic"?) named Value. So I changed my sendablebuilder to provide the heading data with the name Value instead of Heading (what I was using) and that fixed the problem.

Could you elaborate on the term "Topic" as I am not clear on what that is from looking at the API doc. Is it a table entry or a data item inside a table entry?

Crossle86, 2022-11-23:
Data items are called properties as well...

Crossle86, 2022-11-23:
OK. Does topic refer to data item or entries or table sections? Topic would seem a higher level than a data item. Is there some doc or a white paper on NT?

rzblue, 2022-11-23:
A topic is a channel for a single piece of data- This is a new term for NT4, analagous to an "entry" in NT3 and older. 

Here is the docs page for NT4- it's still in development so please make suggestions if you have any.

https://docs.wpilib.org/en/latest/docs/software/networktables/index.html

Crossle86, 2022-11-24:
Ah. Silly me. I was looking at the doc installed on my PC, which is not up to date.

Crossle86, 2022-11-29:
I spoke too soon. After changing the data topic in my sendable to "Value", the gyro display appeared as expected and all seemed good that day. A couple days later I resumed testing and observed the gyro display is gone again, showing "unknown gyro". The sendable shows the data item is still named Value.

PeterJohnson, 2022-11-29:
If this is in the simulation GUI, is a dashboard connected?  If there's no dashboard connected when this happens and connecting a dashboard fixes it, this is the same problem as wpilibsuite/allwpilib#4674.

Crossle86, 2022-11-29:
Yes, connecting outlineviewer causes the gyro display to appear.

closed 2022-11-29 by Crossle86